guysoft/FullPageOS #464: "Xsession: unable to launch [...]" after slpashscreen

BobRubeus, 2023-04-20:
Hi !

I tried to install the last stable version of fullpageos from the Raspberry Pi Imager,
But each time, after every steps of the first boot I got this message : 

```
Xsession: unable to launch "/home/pi/scripts/start_gui" X session ---
"/home/pi/scripts/start_gui" not found; falling back to default session.
```

I have this message even if I modify or not the files on the boot partition.

I tried to install à classic version of Raspbian OS from Raspberry Pi Imager and everything work fine.

Any ideas ?
Thanks by advance ! :)

BobRubeus, 2023-04-20:
I just tried your lastest version available one Rasperry Pi Imager (Nightly) and I have the same issue.

BobRubeus, 2023-04-20:
I found a other solution,
A more logical one.

I downloaded your .zip file and withe the image withe an other software,
Everything work.

I don't know why Raspberry Pi Imager doesn't work.
(on every MicroSD card -old and brand new-, and many different RaspPi)

If you have any ideas, that would be great.

Have a nice day !

redfive2012, 2023-05-18:
I solved it by creating a symbolic link in /home:
`ln -s /home/<real user> /home/pi`
Then I rebooted. It ran the start_gui script, went dark for a minute or 2, then rebooted, and after another couple minutes it loaded my page.

guysoft, 2023-05-18:
I guess we can move the script paths there to a new location outside the user folder, and that should solve it.

BobRubeus, 2023-05-19:
Thanks, I will try !
I don't have a lot of time right now, that can take a long time, by I don't forget you !

marsmaennchen, 2024-02-26:
Hi, i have the exact same problem.
my pi (hostname and user) is named hmi2
in folder /boot the fullpageos.txt:

`http://IP_OF_MY_RASPBERRY:8123/lovelace/default_view`

in the folder /home/hmi2/scripts the start_gui:
`#!/bin/bash

# Stop the screen from turning off when idle.
DISABLE_POWER_MANAGEMENT=yes

# Rotate screen if needed, see 'xrandr -h' for options.
DISPLAY_ORIENTATION=normal

if ["${DISPLAY_ORIENTATION}" != 'normal'];
then
  xrandr --orientation ${DISPLAY_ORIENTATION}
fi

if ["${DISABLE_POWER_MANAGEMENT}" == 'yes'];
then
  xset s off         # don't activate screensaver
  xset -dpms         # disable DPMS (Energy Star) features.
  xset s noblank     # don't blank the video device
fi

matchbox-window-manager &

/home/hmi2/scripts/run_onepageos
`
in the folder /home/hmi2/scripts the run_onepageos:

`#!/bin/bash

while true
do
    if [ $(curl -sL -w "%{http_code}\\n" "$(/home/hmi2/scripts/get_url)" -o /dev/null) == "200" ] || grep -q disabled "/boot/check_for_httpd" ; then
      xdotool mousemove 9000 9000
      /home/pi/hmi2/start_chromium_browser
    fi
    sleep 1
done
`

Does anyone see whats wrong here?
I already read the hint :"I guess we can move the script paths there to a new location outside the user folder, and that should solve it."
But i ont know what that means exactly and how to do that.

guysoft, 2024-02-26:
Please try the latest nightly and update if this fixes your problem

marsmaennchen, 2024-02-26:
i installed now via Raspbery Installer the Nightly 2024-02-23.
1. No more "Webpage can be changed from /boot/fullpageos.txt"
   File doesn't exist. Shall i make one?
2. VNC Viewer doesn't work. Via raspi-config enabled. No connection. In the last build it worked like this.

But at least i have now a Screen with:
[image]

Means i need a Keyboard?

guyshe-jfrog, 2024-02-26:
1 . Huh? thats a bug, it should boot in to chromium with no login screen, I will test again. I tested here before and got the right browser.
2. thats a bug to test, will look in to it.

marsmaennchen, 2024-02-26:
Now it works. It takes an **unusually** long time (about 5 to 7 min) to boot for the **first time**, but after that it works.

guysoft, 2024-02-28:
It does take longer on first boot, because it generates ssh keys and such. It also reboots, so you might be counting that.
I think its around 5 mins. 7 sounds like its taking longer than usual. But if it works it works. You are welcome to send optimizations PR :)

guysoft, 2024-03-25:
Good news - fixed in nightly

closed 2024-03-25 by guysoft