guysoft/FullPageOS #469: How do I get touch to work?

ablohme86, 2023-05-20:
#### What were you doing?
Trying to touch the screen
[Please be as specific as possible here]

#### What did you expect to happen?
The touch to act as a mouse
#### What happened instead?
Nothing
#### Was there an error message displayed? What did it say?
Nope
#### Version of FullPageOS?
2023-05-13_2023-05-03-fullpageos-bullseye-armhf-lite-0.13.0.zip.md5
[Can be found in /etc/fullpageos_version ALWAYS INCLUDE.]

#### Screenshot(s) showing the problem:

[If applicable. Always include if unsure or reporting UI issues.]

#### If you are building FullPageOS - provide a build.log that is created for the build

guysoft, 2023-05-21:
What type of touch screen?
Have you taken a look at this in the wiki?
https://github.com/guysoft/FullPageOS/wiki/PI-3.5-inch-Touchscreen

guysoft, 2023-06-29:
Closing if no reply

closed 2023-07-09 by ablohme86