guysoft/FullPageOS #470: Grafana dashboard not opening on bootup

Khizar-Akhtar, 2023-05-31:
Fullpageos is a great way to open web kiosk and i have using it for many purposes.
but i am facing problem that i am unable to open grafana dashboard on bootup.

### What were you doing?
I am trying to open grafana dashboard on chromium through corporate network on raspberry pi 4b on boot.
 
#### What did you expect to happen?
it should open the grafana dashboard on booting as i have installed the certificates and provided domain name in etc/host file.
I am able to open grafana dashboard on separate tab by pressing ctrl+t

#### What happened instead?
On bootup it is showing black screen with mouse cursor on bootup

#### Was there an error message displayed? What did it say?
 No there was not any error just the black screen with mouse cursor on it

#### Version of FullPageOS?
version is 0.13.0

--------------------------------------
I am able to open grafana dashboard on a separate chromium tab. so i believe it cannot be certificate or domain issue.
But only problem is that it will not open grafana on bootup.

guysoft, 2023-05-31:
Hey, 
1. 
Might it be that the Pi has no internet access? It starts chromium only after it detects an internet connection:
https://github.com/guysoft/FullPageOS/blob/devel/src/modules/fullpageos/filesystem/home/pi/scripts/run_onepageos#L5
2. Did you try setting a different url to see if it works?
3. 0.13.0 has not been released yet, did you try the stable version too?

Khizar-Akhtar, 2023-05-31:
Hi ,
1) Yes it has the internet access , as i am able to open other websites e.g google or youtube.
2). Yes different urls are working on bootup if i change the url from boot/fullpageos.txt file
3) . Yes i tried stable version but in stable version i am unable to install libnss3-tool package ,as i need it to install certificates using certutil . These certificates are necessary to open grafana dash board on a corporate network

guysoft, 2023-05-31:
Not sure how to debug this. I guess you could try a publicly accessible grafana, eg: https://data.octoprint.org/

Khizar-Akhtar, 2023-06-05:
Hi ,Thanks for your reply.

i was able to solve this by commenting out following lines in  home/pi/scripts/run_onepageos

while true
do
#    if [[ $(curl -sL -b cookiefile -w "%{http_code}\\n" "$(/home/pi/scripts/get_url)" -o /dev/null) =~ ^([23][0-9]{2,2}|401)$ ]] || #grep -q disabled "/boot/check_for_httpd" ; then
      xdotool mousemove 9000 9000
      %BROWSER_START_SCRIPT%
  #  fi
    sleep 1
done

Khizar-Akhtar, 2023-06-05:
closing issue

closed 2023-06-05 by Khizar-Akhtar