guysoft/FullPageOS #477: is Pi Zero 2 W supported?

ubeyou, 2023-09-03:
I have a spare Pi Zero 2 W over here and I wasn't sure if Zero 2 W were supported.

Researched a few discussions here, seems like ARMv6 are not supported, but Pi Zero 2 W is using ARMv8.

guysoft, 2023-09-03:
Yes it should work, because its based on RaspberryPi 3.

abd7786, 2023-09-09:
How's the performance for simple website displays? Is the ram sufficient?

guysoft, 2023-09-09:
That depends on the website. Some websites take up lots lf ram abd others far less.
It should work on not too heavy ones. You would have to test or profile the website

joysfera, 2023-10-13:
I am just testing RPi Zero 2 W, seems to work OK.

bluecraank, 2023-11-21:
So my results are very disappointing:
- Very easy sites like google.com or similar (less images, less javascript etc) are working but already with bad performance
- Signage websites (like open-signage) does not work. Zero 2W is dying because RAM overflows extremely

I dont recommend!
Zero 2W seems to have struggle with GUI

cheindel, 2024-05-06:
I am still testing it and have not come to a final verdict, yet. However, it does seem to do the job.

So far I needed to create custom HTML that is not to demanding in the 1 GHz / 512 MB RAM, because otherwise Chrome would show me a sad face instead of the website. Right now it runs a bit warm, showing a simple dashboard based on a Blazor server side app (with an open websocket connection to receive messages to display). 
I
t seems to get through the day and do the job but certainly depends on the use case and the complexity of the dashboard (single webpage vs. full blown web app).