guysoft/FullPageOS #479: Rebooting Pi everymorning

kensaut, 2023-09-21:
I am running a published Google Slide and would need to reboot the Pi or refresh the webpage everyday.  Does this do them by default?  If not, is there a way to do this?

guysoft, 2023-09-21:
You can also just refresh the webpage. There is a script to refresh:
https://github.com/guysoft/FullPageOS/blob/devel/src/modules/fullpageos/filesystem/home/pi/scripts/refresh

You could set it in a crontab

kensaut, 2023-09-21:
> You can also just refresh the webpage. There is a script to refresh: https://github.com/guysoft/FullPageOS/blob/devel/src/modules/fullpageos/filesystem/home/pi/scripts/refresh
> 
> You could set it in a crontab

Does the webpage refresh by default at a certain time?  

Thank you for the information.  If the webpage doesn't refresh, is there a way I can break out of the webpage?  I don't know how to include the script you sent.

guysoft, 2023-09-21:
The time would be set by what ever you put in the crontab. You will have to have some basic knowledge in linux to do this.
I don't understand what "break out of" means in this context.

kensaut, 2023-09-22:
If I can't SSH into the Pi, how could I edit crontab?

guysoft, 2023-09-22:
You can't, you need to have ssh access to the pi or edit the sdcard on a PC directly that can read ext4 file systems.

kensaut, 2023-09-25:
How can I edit the sdcard on a PC to run the script every morning?

guysoft, 2023-09-25:
https://bc-robotics.com/tutorials/setting-cron-job-raspberry-pi/

kensaut, 2023-09-25:
Thank you.

I can't seem to SSH into the Pi or cancel the webpage to do this work.
Thank you for your time.

guysoft, 2024-04-07:
Moving this to discussions