guysoft/FullPageOS #485: splash not showing and boot problems

boatbuddy, 2023-10-27:
#### What were you doing?
booting local webapp. i replaced /boot/splash.png with a png of the same size. 

#### What did you expect to happen?
to see a splash screen when booting. Also boots to black screen often, 20% of the time.
#### What happened instead?
boots to black screen often, 20% of the time, and no splash screen shows at all
#### Was there an error message displayed? What did it say?
no
#### Version of FullPageOS?
2023-05-09_2023-05-03-fullpageos-bullseye-armhf-lite-0.13.0
[Can be found in /etc/fullpageos_version ALWAYS INCLUDE.]

#### Screenshot(s) showing the problem:

[If applicable. Always include if unsure or reporting UI issues.]

#### If you are building FullPageOS - provide a build.log that is created for the build

boatbuddy, 2023-10-27:
it shows the splash screen on initial boot, but every boot after never shows, i did not make any changes except to replacing the splash image, but if i do a fresh install of fullpage os, and do not change the splash at all, same thing, shows splash only on boot 1st time

guysoft, 2023-10-27:
Does it always show the splash screen if you dont change it?
What version of RaspberryPi?
How long does the boot actually take?

boatbuddy, 2023-10-27:
It only shows once, on initial boot after fresh os. On second boot with no modification it doesn't show. Its a raspberry pi 4. Boot takes 30 seconds which is a huge problem I'm trying to fix

guysoft, 2023-10-27:
What are you using it for that the splash screen is such critical?
A workaround might be to have the splash screen load as a background for the GUI.

boatbuddy, 2023-10-27:
Well its not super important,  my thing was because the boot takes so long atleast seeing my logo wouldn't be so bad. Starring at a black screen for 30 seconds is horrible for a digital dashboard for marine

guysoft, 2023-10-27:
Well I am not sure how well FullPageOS was tested for combat.
Understanding the use case might help more here.
Linux as an operating system has verious stages, and in order for you to display a consistent progress it really depends what you are doing.
You could do something simple like adding a wallpaper to the GUI, to something complicated involving recompiling the kernel to show something right at boot

boatbuddy, 2023-10-28:
I have designed a dashboard in html and Javascript,  it will show engine data, rpm etc. It gets its data via serial. Thank you for helping,  I tried doing a headless os install that starts a browser and fullpage os still boots faster. I'm thinking maybe I learn qt as there's alot of videos of a dashboard booting in 5 seconds

Linux as an operating system has various stages, and in order for you to display a consistent progress it really depends what you are doing.

guysoft, 2023-10-28:
5 seconds it the boot time of a bare metal system such as minisynth https://github.com/rsta2/minisynth
It has no linux. You might get cli to boot in about 10. But GUI takes time to load

boatbuddy, 2023-10-29:
Yea, is there anything you could recommend to bring down boot time with fullpage os?

guysoft, 2023-10-30:
You can try looking which service takes time and if its not critical remove it. That can be analyzed using:
```
Systemd-analyze blame
```
https://raspians.com/how-to-fast-boot-raspberry-pi/#Apply_the_Systemd-Analyze_Blame