guysoft/FullPageOS #486: FullPageOs and 7in touch screen touch not working

JecoJetson, 2023-11-16:
Hello, I am having trouble with the raspberry pi 7in touch display. The touchscreen doesn't seem to be registering any touch events. I have tried multiple raspberry pi 4's and checked all my connection and still cant get the touch screen to work. It does seem to work first try using rasbian though.

#### What were you doing?

Have messed around a bunch with different commands and code snippets, especially around the dtoverlay variable in /boot/config.txt but I've had no luck. I've read that Bullseye had issues with the touch capabilities some years ago but should have been fixed already, and these pis were only recently flashed using the imager.

#### What did you expect to happen?

To be able to use the touch screen 

#### What happened instead?

No touch capabilities at all, the display works fine though. it is responsive and works well with a mouse

#### Was there an error message displayed? What did it say?

using dmesg:
```
[    6.843592] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM4345/6 wl0: Nov  1 2021 00:37:25 version 7.45.241 (1a2f2fa CY) FWID 01-703fd60
[    7.751808] uart-pl011 fe201000.serial: no DMA platform data
[    7.859925] rpi_touchscreen_attiny 10-0045: Failed to read REG_ID reg: -5
[    7.859982] rpi_touchscreen_attiny: probe of 10-0045 failed with error -5
[    7.950104] Adding 102396k swap on /var/swap.  Priority:-2 extents:1 across:102396k SSFS
```

#### Version of FullPageOS?

0.13.0

flashed using rpi imager

M0LTE, 2023-11-19:
Mine was working fine - flashed with rpi imager - until I just `apt upgrade`d it. Now the touchscreen is dead. Can still interact with it via VNC, so it's definitely something hardware related.

For now I'll go back to my backup.

guysoft, 2023-11-19:
There is a bug reported here that after updating rpi change some spi settings that break it:

https://www.teamfdm.com/forums/topic/1605-touch-screen-non-reponsive/

From that post:

I did not use the klipper solution as I was unaware of this.

This is what I did and it worked after these steps:

1. SSH into the Raspi -Pi and run

sudo raspi-config

2. Choose interface options:

3, Enable SPI Kernel

4. Enable I2C

5. Reboot

There are images also on that post so you can see how it looks.
Might be worth to test that before @M0LTE you start restoring a backup.

JecoJetson, 2023-11-20:
I solved my problem. Not sure of this is typical but there are 2 dtoverlay variables in config.txt, I kept editing the dtoverlay under [all] vs the one a few linues up under the line 
```
# Enable DRM VC4 V3D driver
```

Commenting out the second dtoverlay and setting the first to fkms fixed my issue.

thanks!

guysoft, 2024-01-15:
Could be added in the wiki FAQ here. closing for now

closed 2024-01-15 by guysoft