guysoft/FullPageOS #487: Adding On screen keyboard to FullPageOS

Linkek, 2023-11-21:
I would like to ask you @guysoft if you want me to add my project [keyboardTriggerServer](https://github.com/Linkek/keyboardTriggerServer) to FullPageOS. Maybe we can make it an optional thing on first "boot"? 

I'm not sure how that will be done. 

Maybe we could talk about that somewhere? Discord?

guysoft, 2023-11-21:
Hey @Linkek 
First thanks for putting the effort in making FullPageOS better! It warms my heart each time I see people building on top of it.

At the very basic I think we could ship a script that you run and does all the install steps.
I saw though there is a manual step in the install process to set the browser in to developer mode and install the extension manually.
I am not sure how you can install an extension automatically on chromium. If you can figure out how to have a script that installs it I'd gladly ship it.

Also if you have any other suggestions would be happy to hear. Keep in mind that I rather not ship nodejs and npm for every installation since most people are not likely to use it.

ShaunThrives, 2024-05-13:
@Linkek 
I have been trying to get this to work to no avail. posted an Issue. Would love to talk to you about it!
@guysoft 
I know it has been around a long time, but cant thank you enough for FullPageOS!
I do have 2 things I need to figure out.
1.  I need a popup keyborad to input user names and passwords
2. I need the OS to ask for WIFI SSID when expected one is not available.
I am not a coder AT ALL but have been muddling through it. If ya have any advise, it would be greatly appreciated!!
Cheers!

guysoft, 2024-06-18:
Hey @ShaunThrives 
1. There are a few projects. take a look at what @Linkek , its special for FullPageOS. There is also a report of another method here https://github.com/guysoft/FullPageOS/issues/489#issuecomment-1999987576 . I have not tried neither of them so I don't know what is better.
2. The question is "how would it ask for an SSID?" FullPageOS is used without a keyboard in most installs, and you can't expect a user to input an SSID. Moreover some installations would consider it a security issue and try to disable any kind of keyboard at all. There might be solutions, but you would need a coder, or find a coder to do them for you (for example you could write your own systemd service, or web ui that talks to NetworkManager).

rapsalands, 2024-12-21:
@guysoft Recently tried running Home Assistant using FullPageOS on a 22 inch touch screen. Its amazing. Thank you for this.

However virtual on-screen keyboard is a total missing element. The above link for Virtual Keyboard is not present in Chrome webstore anymore :(

JamieLaing, 2025-05-08:
Hi @guysoft, just got FullPageOS running on a Pi with this [touchscreen hardware](https://www.waveshare.com/wiki/RPI-HMI-080D-101D) and it works great, but I also need an onscreen keyboard.