guysoft/FullPageOS #489: Virtual Keyboard

McMegabyte, 2023-12-03:
Using a virtual keyboard is not possible. We use FullpageOS to enter text on a touchscreen. Even the latest Chromium didn't offer a solution. Florence cannot be installed either. What can we do? we urgently need help.

guysoft, 2024-01-15:
Hey, what is your usecase? What is FullPageOS running? 
Who is "we" exactly?

csabal, 2024-02-11:
I need this feature too... :( just realized there is no keyboard on touchscreen.

Is there any workaround this?

guysoft, 2024-02-11:
You will need to find a linux component that does this.
You can try to have a look at this:
https://packages.debian.org/search?keywords=onboard

Or there is also ``matchbox-keyboard``, and matchbox is the window manager CustomPiOS uses.

re3Dev, 2024-03-15:
We modified a version of xontab keyboard for a vue.js web application and works much better than all of the alternatives. https://github.com/re3Dprinting/virtual_keyboard