guysoft/FullPageOS #493: Rpi3 Sleeps after 10 minutes with the message No Signal

awasson, 2024-01-01:
This is for a kiosk I've put together for running a simple calendar.

#### What were you doing?
Installed FullPageOS on my SD Card.
Edited fullpageos-wpa-supplicant.txt for local network.
added a localhost site in /var/www/html for the calendar
edited /boot/fullpageos.txt to point to my site. 

#### What did you expect to happen?
I expected my site which dynamically updates itself to run coninuously 

#### What happened instead?
The page displays just fine at boot and updates as normal but 10 minutes after boot, the display reports that it has no signal and goes to sleep. It's a touch screen so it I tap it, the display will come back. 

#### Was there an error message displayed? What did it say?

NO SIGNAL

#### Version of FullPageOS?
Latest 0.13.0

#### Hardware
Raspberry Pi 3
HDMI-Compatible IPS LCD Touch Screen 1024x600 (Amazon)
2.5 Amp switching power supply with on/off switch. 

#### Actions to date
**Edited:** /boot/config.txt

**Uncommented:**
hdmi_force_hotplug=1
config_hdmi_boost=4
**Results:** No change

**Edited:** /etc/rc.local
**Added:** setterm -blank 0 -powerdown 0
**Results:** No change

If this is just a function of the HDMI-Compatible IPS LCD Touch Screen I'm using and an official Rpi Display will work better, I'll ditch this one and get one that works.

Thanks,
Andrew

awasson, 2024-01-01:
I may have resolved this by using sudo raspi-config and disabling Screen blanking under **Display Options** -> **Screen Blanking**

I just adjusted it and although when I returned from raspi-config it complained _dpkg-query: no packages found matching xscreensaver_, it has been up for 17 minutes.

Happy New Year and hopefully tomorrow it will still be up.

Cheers,
Andrew

guysoft, 2024-01-01:
Strange, that should disable power off settings here:
https://github.com/guysoft/CustomPiOS/blob/devel/src/modules/gui/filesystem/home/pi/scripts/start_gui#L14

I can see that the recent changes did in rpi-config are only about blanking if you use wayland:
https://github.com/RPi-Distro/raspi-config/blob/bookworm/raspi-config

So unless wayland was changed to default it should work.

If you can check what is the output of ``XDG_SESSION_TYPE`` that could help determine that

You can get a terminal in the gui by following this:
https://github.com/guysoft/FullPageOS/wiki/FAQ#how-can-i-open-a-terminal-in-the-gui

awasson, 2024-01-02:
I just checked and it doesn't appear to be using wayland echo $XDG_SESSION_TYPE returns tty.

jritsema, 2024-03-03:
Hey, quick question @guysoft.  Curious when/where does this file run on boot?  If I want to customize this, what file would I edit?

guyshe-jfrog, 2024-03-04:
Here:
https://github.com/guysoft/CustomPiOS/blob/devel/src/modules/gui/filesystem/root_init/usr/share/xsessions/guisession.desktop#L4

I am working on it too. I want to move all of that to ``/opt/custompios/scripts``

I started be did not commit it yet.

guysoft, 2024-03-04:
Here:
https://github.com/guysoft/CustomPiOS/blob/devel/src/modules/gui/filesystem/root_init/usr/share/xsessions/guisession.desktop#L4

I am working on it too. I want to move all of that to /opt/custompios/scripts

I started be did not commit it yet.

Kureigu07, 2024-03-28:
Changing start gui to the solution here https://forums.raspberrypi.com/viewtopic.php?t=357209#p2141143 does the trick for me (actually disabling screen blanking).
Any thoughts why this works or how its different?