guysoft/FullPageOS #494: Orca Accessibility

krperry, 2024-01-02:
#### What were you doing?
I want to create a quick  Kiosk and the web page I want to run works perfectly on Full  page OS>  The problem is the web page is for the blind and Visually impared. I myself am a 100% Blind Senior Software Engineer.  I have had someone create the image for me and the web page I want boots up fine and it has some self speaking features so I know it works.  The problem is it also has parts that are spoken by the screen reader of the given OS if you go to it from Windows, Linux, and mac.  Unfortunetly if I want to set this up with Full page OS.  There is no way that I know of yet to turn on Orca which is the screen reader under Linux.  I am sure there is a way to do this so that a blind person can both set this up and so that the Full Page OS talks with a screen reader.    I will be looking at the compile script and seeing if I can add in the accessibility.  If someone is wizar and can add this as a variant using an argument to build this please let me know and make it part of the read me.
[Please be as specific as possible here]

#### What did you expect to happen?

#### What happened instead?

#### Was there an error message displayed? What did it say?

#### Version of FullPageOS?

[Can be found in /etc/fullpageos_version ALWAYS INCLUDE.]

#### Screenshot(s) showing the problem:

[If applicable. Always include if unsure or reporting UI issues.]

#### If you are building FullPageOS - provide a build.log that is created for the build

guysoft, 2024-01-03:
That's a pretty cool use case. 

There is a way to get Orca working on Linux, and RaspberryPi in particular.
This guide takes you though it. You don't have to do the first part of it, ignore the mate desktop part, it gets relevant from how to compile it, to how to set up X and even chromium which is what FullPageOS uses:
https://techesoterica.com/getting-the-orca-screen-reader-working-on-a-raspberry-pi-4-with-raspbian-buster-and-the-mate-desktop/

Are you building this as a one-off single install? Or you are planning to have this work on multiple devices and you want to maintain a build?
If its just one device you could compile Orca on the pi directly, then configure the X screen settings there and chromium.

```
git clone https://gitlab.gnome.org/GNOME/orca.git # you might want to check out a tag, not just the master/main branch
sudo apt-get build-dep gnome-orca
sudo apt-get update; apt-get install checkinstall
 cd orca
PYTHON=/usr/bin/python3 ./autogen.sh
make
sudo checkinstall
```

(set up the dummy screen as mentioned in the guide).

To make chromium start wtih ``--force-renderer-accessibility`` edit on FullPageOS the file:

https://github.com/guysoft/FullPageOS/blob/devel/src/modules/fullpageos/filesystem/home/pi/scripts/start_chromium_browser

You can add the flag there (The flags might be different in older versions, the neat list is recent)

krperry, 2024-01-03:
This is going to be an image we put on Raspberry PI 400  and it is an image we will have to hopefully give / sell to a lot of people.  The software is running under javascript what we would like to do is make it so it does not need wi-fi.   So the plan would be to Add our software to the device as well as turn on accessibility  This way they would not have to set the web page the web page would default our application.  I need to look at the script to create the image and find out how to put our web application in as a default installed application as well as adding the screen reader.  

 

https://techesoterica.com/getting-the-orca-screen-reader-working-on-a-raspberry-pi-4-with-raspbian-buster-and-the-mate-desktop/

https://github.com/guysoft/FullPageOS/blob/devel/src/modules/fullpageos/filesystem/home/pi/scripts/start_chromium_browser

guysoft, 2024-01-03:
You can change the default web application from ``/boot/fullpageos.txt`` [as written in the readme](https://github.com/guysoft/FullPageOS/issues/494)

You will also need to disable the check that starts chromium browser only when the internet comes online (the common scenario is people want internet before the webpage loads):
https://github.com/guysoft/FullPageOS/blob/devel/src/modules/fullpageos/filesystem/home/pi/scripts/run_onepageos#L5
You can disable the check by creating a file with the content "disabled" in ``"/boot/check_for_httpd"``

krperry, 2024-01-03:
Is there a specific place I could put an javascript application

Or do I have to  add our application to the bbuild script?  I guess I could install orca, install our application and then save the image off the card to make this easy.

 

https://github.com/guysoft/FullPageOS/blob/devel/src/modules/fullpageos/filesystem/home/pi/scripts/run_onepageos#L5

guysoft, 2024-01-04:
Does this javascript application require a backend such as nodejs or just javascript?
If its just javascript then FullPageOS shipps with apache. You can place it at ``/var/www/html``
Similar to how [its done here in the build process](https://github.com/guysoft/FullPageOS/blob/devel/src/modules/fullpageos/start_chroot_script#L53)

krperry, 2024-01-04:
It will need node js so I am assuming I will need to install that.  I am going to work on this this weekend.

guysoft, 2024-01-04:
If you need nodejs then you need to install it and do all that, then you can add a systemd service that would start it