guysoft/FullPageOS #497: Update image in Raspberry Imager

nikolaev-rd, 2024-01-05:
Last stable release of FPO was in 2023-10-22, but in Imager now is very old image:
[image]
Nightly builds also outdated.

How I can help to update it?

guysoft, 2024-01-05:
Latest stable was released on 2023-10-22 (tagged). but it was built on 2022-03-07. As you can see in the release:
https://github.com/guysoft/FullPageOS/releases/tag/0.13.0

The nightly builds are built on demand (I press a "build" button on the Jenkins instance that builds).

I just started a nightly build now so you have something more recent.

There was an RC out ages but it got stuck since I was really busy in my life You can see it took a while here:
https://github.com/guysoft/FullPageOS/issues/412#issuecomment-1774168145

If you want to help update it then really testing and confirming an RC is good enough is what is missing. No on on the latest RC said "this is good, I am happy with it". which meant I didn't know if its OK to release. I could make a nightly, make an RC issue and we could test it to confirm its good enough to release and have 0.14.0.

nikolaev-rd, 2024-01-07:
Ok, so I'm waiting from your Nightly Release link to test it o PRi2, or maybe I can already download it from somewhere?

nikolaev-rd, 2024-01-07:
Don't known your release process, but maybe we can create some pre-release candidates right here, on github? So anybody can download & test it...

guysoft, 2024-01-08:
Hey @nikolaev-rd.
So the process today is that I run a jenkins build on my own selfhosted server that I maintain.
That gets uploaded here (along with the build log):
https://github.com/guysoft/FullPageOS#where-to-get-it

When a good build is out I publish an issue, for example this one:
https://github.com/guysoft/FullPageOS/issues/412#issuecomment-1774168145
The next one would have the format:
``FullPageOS 0.14.0 RC1 status``

And also the image would get uploaded to rpi-imager using this script:
https://github.com/guysoft/unofficialpi/blob/main/src/create_rpi-imager-distro.py

Then you and others can test it.

Recuarding moving to github actions.
I was looking at that route. The issue is that its harder to maintain on multiple images I maintain.

There is already a github action workflow run here:
https://github.com/guysoft/FullPageOS/actions/runs/7414322544
This is the gitub action:
https://github.com/guysoft/FullPageOS/blob/devel/.github/workflows/main.yml

It might work if someone gives it more love and care than I have the time.

nikolaev-rd, 2024-01-09:
I tested this one: [2023-05-12_2023-05-03-fullpageos-bullseye-armhf-lite-0.13.0.zip](http://unofficialpi.org/Distros/FullPageOS/nightly/2023-05-12_2023-05-03-fullpageos-bullseye-armhf-lite-0.13.0.zip) — works on PRi 2.

nikolaev-rd, 2024-01-12:
Just tested [2024-01-12_2023-12-11-fullpageos-bookworm-armhf-lite-0.14.0.zip](http://unofficialpi.org/Distros/FullPageOS/nightly/2024-01-12_2023-12-11-fullpageos-bookworm-armhf-lite-0.14.0.zip) on RPi 4, and there is no file `/boot/fullpageos.txt`:

```
ls -la /boot/
total 71482
drwxr-xr-x  3 root root     4096 Dec 11 04:26 .
drwxr-xr-x 18 root root     4096 Dec 11 04:26 ..
-rwxr-xr-x  1 root root      169 Dec 11 04:26 cmdline.txt
-rw-r--r--  1 root root   200207 Nov 24 19:11 config-6.1.0-rpi7-rpi-v6
-rw-r--r--  1 root root   205495 Nov 24 19:11 config-6.1.0-rpi7-rpi-v7
-rw-r--r--  1 root root   223069 Nov 24 19:11 config-6.1.0-rpi7-rpi-v7l
-rw-r--r--  1 root root   230405 Nov 24 19:11 config-6.1.0-rpi7-rpi-v8
-rwxr-xr-x  1 root root     1259 Dec 11 04:26 config.txt
drwxr-xr-x  5 root root     6144 Jan  1  1970 firmware
-rw-r--r--  1 root root 10475286 Dec 11 04:24 initrd.img-6.1.0-rpi7-rpi-v6
-rw-r--r--  1 root root 10585737 Dec 11 04:24 initrd.img-6.1.0-rpi7-rpi-v7
-rw-r--r--  1 root root 10682270 Dec 11 04:24 initrd.img-6.1.0-rpi7-rpi-v7l
-rw-r--r--  1 root root 10588540 Dec 11 04:24 initrd.img-6.1.0-rpi7-rpi-v8
lrwxrwxrwx  1 root root       18 Dec 11 04:24 issue.txt -> firmware/issue.txt
lrwxrwxrwx  1 root root       17 Dec 11 04:06 overlays -> firmware/overlays
-rw-r--r--  1 root root       83 Nov 24 19:11 System.map-6.1.0-rpi7-rpi-v6
-rw-r--r--  1 root root       83 Nov 24 19:11 System.map-6.1.0-rpi7-rpi-v7
-rw-r--r--  1 root root       83 Nov 24 19:11 System.map-6.1.0-rpi7-rpi-v7l
-rw-r--r--  1 root root       83 Nov 24 19:11 System.map-6.1.0-rpi7-rpi-v8
-rw-r--r--  1 root root  6694832 Nov 24 19:11 vmlinuz-6.1.0-rpi7-rpi-v6
-rw-r--r--  1 root root  7023344 Nov 24 19:11 vmlinuz-6.1.0-rpi7-rpi-v7
-rw-r--r--  1 root root  7464952 Nov 24 19:11 vmlinuz-6.1.0-rpi7-rpi-v7l
-rw-r--r--  1 root root  8760390 Nov 24 19:11 vmlinuz-6.1.0-rpi7-rpi-v8
```

CamDavidsonPilon, 2024-01-12:
Try `ls -la /boot/firmware` (it changed with bookworm)

nikolaev-rd, 2024-01-12:
> Try `ls -la /boot/firmware` (it changed with bookworm)

yes, found it. thanks!

nikolaev-rd, 2024-01-12:
It seems like something went wrong:
`/home/pi/scripts/run_onepageos`
```
head: cannot open '/boot/fullpageos.txt' for reading: No such file or directory
grep: /boot/check_for_httpd: No such file or directory
head: cannot open '/boot/fullpageos.txt' for reading: No such file or directory
grep: /boot/check_for_httpd: No such file or directory
head: cannot open '/boot/fullpageos.txt' for reading: No such file or directory
grep: /boot/check_for_httpd: No such file or directory
```

Error because in script `/home/pi/scripts/run_onepageos`:
```bash
#!/bin/bash

USER_AGENT="Mozilla/5.0 (X11; Linux x86_64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/120.0.0.0 Safari/537.36"

while true
do
    if [[ $(curl -sL -b cookiefile -w "%{http_code}\\n" -H "user-agent: ${USER_AGENT}" "$(/home/pi/scripts/get_url)" -o /dev/null) =~ ^([23][0-9]{2,2}|401)$ ]] || grep -q disabled "/boot/check_for_httpd" ; then
      xdotool mousemove 9000 9000
      /home/pi/scripts/start_chromium_browser
    fi
    sleep 1
done
```
used script `/home/pi/scripts/get_url`:
```bash
#!/bin/bash

SERIAL=`cat /proc/cpuinfo | grep -i '^Serial' | awk '{ print $3 }'`
URL="$(head -n 1 /boot/fullpageos.txt | sed -e "s/{serial}/${SERIAL}/g")"
echo $URL
```

And yes, file `/boot/fullpageos.txt` doesn't exist. Workaround:
```shell
ln -s /boot/firmware/fullpageos.txt /boot/fullpageos.txt
```
Result:
```shell
ls -l /boot/fullpageos.txt
lrwxrwxrwx 1 root root 29 Jan 12 22:02 /boot/fullpageos.txt -> /boot/firmware/fullpageos.txt
```

guysoft, 2024-01-13:
I'll add a fix for that

nikolaev-rd, 2024-01-13:
[This workaround](https://github.com/guysoft/FullPageOS/pull/502/files) don't work: boot till splash screen and then reboot again

Speedymatze, 2024-01-14:
Hello, it's still not working for me. Is there a new version?

nikolaev-rd, 2024-01-14:
> Hello, it's still not working for me. Is there a new version? Black screen and the cursor flashes wildly.

Did your try the [full script](https://github.com/guysoft/FullPageOS/pull/502/files) as workaround? It's worked for me (PRi 4):

```bash
FULLPAGEOS_BOOT_FILES=(
"fullpageos.txt"
"fullpagedashboard.txt"
"check_for_httpd"
"splash.png"
)
for FILE in "${FULLPAGEOS_BOOT_FILES[@]}"
do
    [ ! -e "/boot/${FILE}" ] && sudo ln -s "/boot/firmware/${FILE}" "/boot/${FILE}"
done
```

Speedymatze, 2024-01-14:
I've just tried it again, but it's always the same result on the RPi 5, but I'm not that familiar with it, so I'm sure I haven't made any mistakes. If I could get a complete image, I would test it again on the RPi 5.

nikolaev-rd, 2024-01-15:
I don't have RPi 5, don't know how to investigate you problem. I think @guysoft can create test build with my fixes — maybe it fix you problem.

Speedymatze, 2024-01-15:
I tried that, but I can't manage to create an image. I have never done that before.

pauldogg, 2024-02-04:
As of today:
Hashes do not match on the raspberry pi imager. All attempts end with a corrupted file warning after flashing.

guysoft, 2024-02-04:
@pauldogg thinks for reporting will look in to it

guysoft, 2024-02-04:
@pauldogg Works for me on the stable and nightly - what are you testing? what version of fullpageos and what version of rpi imager?

guysoft, 2024-02-04:
Ok reproduced, will make a new nightly that should fix it.

guysoft, 2024-02-19:
Fixed, I also updated the CI/CD to handle this better.
It indicates if the upload worked correctly and puts all the process in one place 
(that's also open here https://github.com/guysoft/unofficialpi/blob/ci/src/ci/Jenkinsfile.custompios)

closed 2024-02-19 by guysoft