guysoft/FullPageOS #499: Keyboard

timboh72, 2024-01-07:
Good morning, I have found FullPageOS through the Pi imager tool and it will be good to allow someone with disability to use without the ability to access the file system and be locked into one or two pages only and therefore not delete anything.
Whilst this is probably a Linux question I would appreciated if there is any way to reenable the keyboard and mouse through the config.txt or similar so the user can still interact with the browser.
My Linux knowledge is limited and I have a way to go to setup a hardened PiOS with limited access.
Thank you in advance

guysoft, 2024-01-15:
Keyboard and mouse are enabled in FullPageOS by default. You can click on links and type. What are you tying to do where they don't work?

LennartSea, 2024-03-11:
I have a similar problem. Mouse and keyboard works but where to xonfigure keyboard type ? 
I need Nordic type.

guysoft, 2024-03-11:
@LennartSea 
rpi-config, it Rpi OS based:
https://linuxconfig.org/how-to-change-keyboard-layout-on-raspberry-pi

I am turning this to a discussion