guysoft/FullPageOS #500: Splash screen only working for 1 frame and going into idle mode after a short while

coreyfreeman82, 2024-01-09:
#### What were you doing?

I installed FullPageOS this morning using the raspberry pi imager. Then I edited fullpageos-wpa-supplicant.txt to add my wifi credentials. I also added a new splash.png and deleted the previous one. Then I put the url which needs to be launched in fullpageos.txt (this works without problems)

#### What did you expect to happen?

A fullscreen kiosk mode view of my url which stays on forever and a splash screen which is shown during reboot.

#### What happened instead?

Raspberry pi goes into idle mode after a short time period (I am able to ssh into it but the monitor shows no signal. My splash screen image is only viewed for 1 frame (like 0.2s) and I can see all the terminal commands that are being executed instead of the splash screen.

#### Version of FullPageOS?

0.13.0

#### Output of file /boot/splash.png 

splash.png: PNG image data, 1920 x 1080, 8-bit/color RGBA, non-interlaced

guysoft, 2024-01-13:
Hey, did you try this on more than one screen? Is this the same behavour on more than one?!

coreyfreeman82, 2024-01-15:
@guysoft Yes I tried it on multiple monitors and they all have the same problems

Extremity, 2024-01-20:
I'm having the blank screen issue, too. Everything takes forever to start - like, 6 minutes. Eventually, though, I do see my page. It works great for about 5 minutes, and then the screen goes black and monitor reports no signal. Pi is still accessible via SSH. Happens on multiple monitors. I've tried the various config options for HDMI signals. 

Pi is on a VLAN with local only, no internet, but the page I have it displaying is at a local IP, so it doesn't require internet access - and like I said, it loads it fine and displays it for a bit.

guysoft, 2024-01-20:
@Extremity What version of RaspberryPi?