guysoft/FullPageOS #503: Set up wifi on first start

Githamza, 2024-01-17:
I'm looking a solution to my use case. 
I would like to integrate fullpageos to tablets then delivers them to users. The issue is that I don't have access to wifi credentials. User should connect the WiFi. 
How can I do this ? Is it possible to configure WiFi on first run or edit WiFi later but directly on tablet ?

guysoft, 2024-01-17:
FullPageOS runs on rasperrypis and not tablets - so just making sure you know that.

There is no browser-ready network config that I know of.
Zynthian team implemented it for their app (that is built on CustomPiOS):
https://github.com/zynthian/zynthian-webconf/blob/testing/lib/wifi_config_handler.py

You could set it up a network manager taskbar that has the nm-applet and then use that, but it would be outside the browser

Githamza, 2024-01-17:
@guysoft  thank you.

So if I would buy a "tablet" that when turned on,  only opens a webpage, how can I do it ? is there kit of touchscreen + raspberry pi + case mounted and ready to use ?

davpirelli, 2024-04-04:
> @guysoft  thank you.
> 
> 
> 
> So if I would buy a "tablet" that when turned on,  only opens a webpage, how can I do it ? is there kit of touchscreen + raspberry pi + case mounted and ready to use ?

If your purpose is to show a web page on a tablet, maybe this is not for you. Moreover, the costs are much higher for the hardware than some Chinese mid-high-end tablets (80/90€). You can use a solution like FullyKiosk Browser on Android, I use it for tablets within some B&Bs.

guysoft, 2024-04-04:
I think its a different solution. A tablet runs on batteries and has different connectivity than a single board computer. Indeed I am not sure what you are trying to do, it might be better using a raspberrypi and it might not.