guysoft/FullPageOS #504: Blank Screen After random times

madakuoni, 2024-01-18:
I use a Raspberry Zero W 2 and everything works Great, but After some random time, the hdmi Output stops. The raspberry is online and i can still use SSH. But on the screen it is Said the hdmi Signal stopped. Is there anything i can Check to find the Problem ?

Extremity, 2024-01-20:
Exact same thing is happening to me. I found this earlier today and it's perfect for my needs, but unfortunately it doesn't look like it's going to work, and the main site for the project hasn't been updated in many years. I've been scanning error logs for the past two hours trying to figure it out.

wolfgangmo, 2024-01-23:
Same here. Raspberry Zero W 2 tops HDMI output after 8 to 12 minutes. Tried different monitors -> no signal. No chance to wake it up.

wgray4431, 2024-02-29:
Have the same issue with a fresh and updated install on a RPi  4 . Installed updated and configured yesterday, it would randomly be a blank screen. Sometimes it would last an hour or two and other times it would be just under 30 minutes and would have the blank screen. Today I plugged it in this morning in the HDMI 0 slot and it has been running for most of the day.
 
Linux fullpageos 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023 aarch64 GNU/Linux

felipog, 2024-03-14:
Having the same issue here.
Latest FullpageOS , tried both variants (normal and no-acceleration variant), with and without all updates, with different pages and even the default page from FullpageOS.

Output stops working after random number of minute, no logs on /var/logs that I could link to this issue.
It just stops outputing.
SSH into it and reboot is the only choice at the moment (do not have mouse connectivity), is in a remote area.
Will try using the VNC to see if restores it too.

Extremity, 2024-03-16:
I switched to a completely custom setup using a few commands to load/refresh/change URL and it's perfect. I've got some HAOS automations set to change the displayed page as needed, and because it doesn't rely on any third party packages I never have to worry about issues like this. I'd be happy to give this a try in the future if it ever gets fixed, though, it would certainly be easier to set up for a first timer.

StevenChisholm, 2024-05-19:
I had this problem on the Zero 2W and the 3B+. Tried a handful of things and couldn't get it working. Finally, I think I have it working. Starting from a Vanilla stable image (version 0.13.0). I configured the WLAN and the Website. Then I SSH. Run `sudo nano /etc/lightdm/lightdm.conf` Find the this line `#xserver-command=X` under `[Seat:*]` and replace it with `xserver-command=X -s 0 -dpms` Save then reboot. Been working for over an hour on the 3B+. Previously it would blank after ~10 minutes.

guysoft, 2024-05-27:
That would suggest that xset does not work,this is in the start script"

```
  xset s off         # don't activate screensaver
  xset -dpms         # disable DPMS (Energy Star) features.
  xset s noblank     # don't blank the video device
 ```

Source:
https://github.com/guysoft/CustomPiOS/blob/devel/src/modules/gui/filesystem/opt/scripts/start_gui#L17

Perhaps a different method will yield better result, take a look at this:
https://forums.raspberrypi.com/viewtopic.php?p=1478379&sid=352526a89f1a4231a78ae441e71233cf#p1478379

russinnes, 2024-06-20:
> I had this problem on the Zero 2W and the 3B+. Tried a handful of things and couldn't get it working. Finally, I think I have it working. Starting from a Vanilla stable image (version 0.13.0). I [...]

Bingo - Same experience for anyone else facing this. Worked like a charm.

guysoft, 2024-06-20:
@russinnes What version of FullPageOS? Should we add it to the latest build?

russinnes, 2024-06-20:
> @russinnes What version of FullPageOS? Should we add it to the latest build?

0.13.0

guysoft, 2024-06-21:
@russinnes Try 0.14.0 RC1, a bug for that was fixed there.