guysoft/FullPageOS #505: Not able to load a locally stored html file.

TeckPete, 2024-01-22:
#### What were you doing?
Trying to open a local stored html page 
[Please be as specific as possible here]
I saved my the html file in /var/www/html/index.html and changed the fullpageos.txt to http://var/www/html/index.html
#### What did you expect to happen?
to pull the the locally stored html page
#### What happened instead?
the screen is black. when a put a web url in it works fine
#### Was there an error message displayed? What did it say?
no error
#### Version of FullPageOS?
0.13.0
[Can be found in /etc/fullpageos_version ALWAYS INCLUDE.]

#### Screenshot(s) showing the problem:
[image]
[image]

[If applicable. Always include if unsure or reporting UI issues.]

#### If you are building FullPageOS - provide a build.log that is created for the build

TeckPete, 2024-01-26:
@guysoft could this have anything to do with my problem? I'm sure I just missed something stupid.

[image]

Metatron2k2, 2024-01-29:
In order to load a local file you would need to use file:// instead of http. Although I am not sure if that resolves you problem. As far as I know chrome disables JS for local files. Therefore you will probaly have to isntall a webserver (apache, nginx lighttp or similar) on the pi

guysoft, 2024-01-31:
There is an onboard websever, but then you need to use https://localhost

https://en.wikipedia.org/wiki/Localhost

I am closing this because its not a FullPageOS issue.

Makes me think we might need a forum