guysoft/FullPageOS #506: FullPageOS goes to black after site refresh

1poc, 2024-01-22:
#### What were you doing?

[Please be as specific as possible here]

#### What did you expect to happen?

#### What happened instead?

#### Was there an error message displayed? What did it say?

#### Version of FullPageOS?

[Can be found in /etc/fullpageos_version ALWAYS INCLUDE.]

#### Screenshot(s) showing the problem:

[If applicable. Always include if unsure or reporting UI issues.]

#### If you are building FullPageOS - provide a build.log that is created for the build

1poc, 2024-01-22:
Hi. im quite new to this but ill try to explain :) 

I have set up FullPage OS and with a URL that works fine. But after some minutes (the website has autorefresh) the screen will just go to black.. What could be wrong?

guysoft, 2024-01-22:
1. Can you provide the url?
2. what version of FullPageOS?
3. What version of Pi?
4. is it a power save issue of the monitor? if you plug in a mouse and move it does the screen go back? Or do you see some kind of error?

1poc, 2024-01-22:
> 1. Can you provide the url? this one f.ex: https://infoplakat.no/osjen-otell-/resepsjon
> 2. what version of FullPageOS? 2022-01-28-fullpageos-bullseye-armhf-lite-0.13.0
> 3. What version of Pi? 4 mod B
> 4. is it a power save issue of the monitor? if you plug in a mouse and move it does the screen go back? Or do you see some kind of error? Is it possible that the raspberry goes to sleep? i can see the screen is on but cant se webpage anymore. If i connect mouse and press/move a bit i get webpage back..

guysoft, 2024-01-22:
The page looks fairly normal, not sure why it would go blank - the question is if the browser crashes somehow or something like that. But there is a recovery for that. So I am not sure if its the screen or the software at this point - can you try a different screen or monitor?

guysoft, 2024-01-24:
Ok, just as a test, if you flash a brand new FullPageOS. And dont change the url, does it still happen?

Again, trying to rule out. If that is the case then its much more likely a FullPageOS system-wide issue, or a hardware issue. Rules out some modification.

guysoft, 2024-03-31:
Hey, the latest nightly should have a background, so if you try that, the screen should go either blank, or to the background image. That would make it possible to detect if the issue is with the screen blanking or the browser crashing.

umpirsky, 2025-02-18:
I have the same problem with `https://www.google.com/` url. Not sure if it's auto refresh, but on fresh install, I just configured `https://www.google.com/` in `/boot/fullpageos.txt`, it works for a few minutes and then monitor just stops recieving the signal. I still see `chromium-browse` in the list of processes via ssh, but see no signal on hdmi port.

umpirsky, 2025-02-18:
Tried few other sites, same issue. I don't think refresh is the problem.