guysoft/FullPageOS #508: Chromium not starting when hostname is changed

klenzel, 2024-01-29:
#### What were you doing?

Change Hostname in /etc/hostname

#### What did you expect to happen?

System is running fine with new Hostname

#### What happened instead?

Chromium is not starting. Instead, a black blank screen is displayed.
If i turn back the Hostname to "fullpageos" an restart the system, the chromium-browser comes back.

#### Was there an error message displayed? What did it say?

`dpkg-query: no packages found matching bluealsa
[16685:16685:0129/042154.956623:ERROR:process_singleton_posix.cc(353)] The profile appears to be in use by another Chromium process (3746) on another computer (fullpageos). Chromium has locked the profile so that it doesn't get corrupted. If you are sure that no other processes are using this profile, you can unlock the profile and relaunch Chromium.
[16685:16685:0129/042154.956957:ERROR:message_box_dialog.cc(146)] Unable to show a dialog outside the UI thread message loop: Chromium - The profile appears to be in use by another Chromium process (3746) on another computer (fullpageos). Chromium has locked the profile so that it doesn't get corrupted. If you are sure that no other processes are using this profile, you can unlock the profile and relaunch Chromium.
`

#### Version of FullPageOS?

2023-12-11-fullpageos-bookworm-armhf-lite-0.14.0
Same with stable Version.

Skifi, 2024-01-31:
Hello, I had same problem.
Try to google SingletonLock in chromium.
I change hostname by script and this line solved my problem:
rm -rf /home/pi/.config/chromium/SingletonLock

sense-design, 2024-02-13:
Can confirm
`rm -rf /home/pi/.config/chromium/SingletonLock`
solves the problem for me too

guysoft, 2024-04-07:
@sense-design we can add that fix to the wiki if you want.

sense-design, 2024-04-07:
Would be great, shall I create the change and do a pull request?

guysoft, 2024-04-08:
@sense-design Sure, I think you can just edit the wiki. But if you only have access to a PR that could work too

sense-design, 2024-04-09:
@guysoft, wiki change added.