guysoft/FullPageOS #509: Sometimes blackscreen after boot, no chromium start

bluecraank, 2024-01-29:
#### What were you doing?

Starting fullpageos on rpi4 4gb

#### What did you expect to happen?

Chromium starts and shows webpage

#### What happened instead?

Blackscreen

#### Was there an error message displayed? What did it say?
No, just blackscreen
#### Version of FullPageOS?
0.14.0

bluecraank, 2024-01-31:
Hello,

i forgot to mention that i have wrote disabled in /boot/check_for_httpd

Its also very random, with and without ethernet connected its happening

pauldogg, 2024-02-04:
No matter what I try (different pis: 3, 3b+, 4, ethernet, wifi...); I cannot get past the black screen with a cursor.
I'm able to SSH and VNC; but nothing else seems to work as expected.

version: 2024-01-12 08:10

Also fullpageos logo fails every time as well.

please help.

thanks

guysoft, 2024-02-04:
Ok, you gave an important bit of information - you have a cursor. That means that the UI is on, but the browser is not starting.
That is really likely to be something with the internet.

Lets try to override everything just to see you got that bit right:
 
 
 Try to replace:
 ```
mv /home/pi/scripts/run_onepageos /home/pi/scripts/run_onepageos.bak
echo "#!/bin/bash" >> /home/pi/scripts/run_onepageos
echo "xterm" >> /home/pi/scripts/run_onepageos
chmod 755 /home/pi/scripts/run_onepageos
```
Then reboot

This should back up the chroium start script. and when you boot you should get an xterm. Its not a browser yet, but at least it means you will not get a totally black screen, and you can try to run commands from there.

Then I would suggest running:

```
/home/pi/scripts/start_chromium_browser
```
That should start chromium with the settings. And if it fails you might get an output on the shell so we can understand what is going wrong.

csabal, 2024-02-10:
HI

2023-05-12 07:34 version seems works, but sometimes it is not. :) 
Ok, so it seems to me, that issue around the network or similar. If the PI builds up the WIFI connections too slow, it is stops with the cursor, so no webpage loading. I think it needs a little delay in startup, so there would be enough time to connect to the WIFI.

How can I add a little start delay? No much experience w Raspberry yet. Thx.

bluecraank, 2024-02-10:
Yeah start delay should help but i dont know where the autostart ist of onepageos

Our pis are connected via wired ethernet

chasecee, 2024-02-15:
I was having this same error so I built my own: https://github.com/chasecee/pi-kiosk-scripts
It's not a drop in replacement, only basic functionality, but could be a guide if you were a frustrated individual like me.

guysoft, 2024-02-18:
I wonder if we should add  a timeout to waiting for internet. so at lease people get something.

guysoft, 2024-02-24:
Recommend you try the latest nightly, lots of stuff was fixed

fladdie, 2024-03-05:
Hi, sorry for the question. Can you explain how I can update FullpageOS via ssh without having to reflash everything to the SD card?
My device is unfortunately 400km away and currently only shows a black page with a mouse pointer.

guysoft, 2024-03-05:
@fladdie Please post questions like this in the discussion forum and not in a random issue that some has posted.

In short - you can't there is no remote update flashing in FullPageOS. And I don't have the capacity to design and build such a thing. In general when you build such a system and you want remote updates you need something such as a two-parition system where you can update one and fallback on the other. Something like this: https://sbabic.github.io/swupdate/overview.html#double-copy

guysoft, 2024-04-07:
Hey, please try the latest RC that should fix the issue.
https://github.com/guysoft/FullPageOS/issues/540

guysoft, 2024-06-05:
Closing if no reply, issue should be fixed in 0.14.0 RC1