guysoft/FullPageOS #510: FullpageOS is fundamentally broken

Slos, 2024-02-02:
#### What were you doing?

configuring fullpageos.txt, testing different websites and looking at which ones load and which ones don't (the one I need to load does not work).

https://www.google.com -> this loads OK
https://www.github.com -> this loads OK

https://www.bbc.com/news -> this does not work....!!??
https://www.standaard.be -> this does not work....!!??
https://www.theguardian.com/international -> this does not work....!!??

my website which I am working on for a client --> also does not load...!

I downloaded an older image of fullpage OS (2018 build I think) and that one works. Who broke this in the mean time and made this project totally useless? 

#### What did you expect to happen?

I expected a browser to start up in fullscreen / kiosk mode and I was hoping the see the webpage from the URL that I configured.

Can someone, who is smarter than me when it comes to this, tell me what is the point of fullpageos if half the URL's don't work? 

All we want is:

1. write the image to the SD card
2. configure URLs to show
3. place the card in a raspberry and turn it on
4. relax

Why can't we have this when we used to have this a few years ago?
 
#### What happened instead?

a black screen

#### Was there an error message displayed? What did it say?

No

#### Version of FullPageOS?

Fullpage OS (stable) from 2022-03-07 (via raspberry pi image writer)

#### Screenshot(s) showing the problem:

N/A

#### If you are building FullPageOS - provide a build.log that is created for the build

guysoft, 2024-02-03:
Hey @Slos 
0. Who is "we"?

Also, if you want an older version they are available for download here.

1. When you get a blank screen - Is this after the boot sequence (boot lines or bootspalsh?) Because if its completely blank it would not matter what website you configure. Its not even reaching that stage.

2. FullPageOS tests if it can connect to website before switching on the browser. So it does not load a broken page. A blank screen might mean its not managing to preform that test. You can try to disable that by creating a file at ``/boot/check_for_httpd`` and see if that helps.

Also - I am working on this software 100% in my free time. I have no clients who pay me for it. If you want to help develop it to become even more stable I welcome contributions and pull requests. I hope you figure out how to fix the issue. When you do I kindly reminder that there is a donation button in the readme file.

guysoft, 2024-02-04:
Hey, trying to link this with #509 
When it does not start - do you have a cursor?

Slos, 2024-02-05:
Hey @guysoft,

Thanks for getting back to me. I very much appreciate you taking your time to respond to issues and for keeping this repository going in your free time.

To answer your first question as to "Who is we?". I'm more than happy to help get to the bottom of the issue at hand. I just need to set up an environment where I can see what exactly is going on by looking at some log files while replicating the issue. On the other hand, whoever made the breaking change should be involved in getting things back on track but I'm afraid we're way past that point unfortunately. 

1. When do I get the blank screen? This is after the boot sequence... probably when it is trying to load the browser and display the page. 

2. I will try your suggestion of disabling 'check_for_httpd' in the config. I was not aware of the behavior you're describing where it first tries to connect to the site and only then launches the browser. Is there a good rationale for this workflow? I'd say launch the browser ALWAYS and let the browser connect to the site. This way if there is an error with the connection, at least there will be some visual feedback on the screen (certificate invalid, no internet connection... etc). At the moment, with the black screen, the user has no clue what it going on.  Am I missing something (or many things)?

Cheers,

Slos

Slos, 2024-02-05:
> Hey, trying to link this with #509 When it does not start - do you have a cursor?

I did see a cursor if I remember correctly (to be honest, I did not look for these kind of details). I will do a few more tests and let you know.

jkriegshauser, 2024-02-05:
I'm running into this too sometimes. Are there any logs that we can check?

jkriegshauser, 2024-02-08:
Another clue from this morning. Blank screen, no cursor. I logged in via ssh and tried to run this, but apparently X wasn't running:
```
pi@mm-pi:~/scripts $ ./start_chromium_browser
dpkg-query: no packages found matching bluealsa
[1335:1335:0208/074359.754701:ERROR:ozone_platform_x11.cc(239)] Missing X server or $DISPLAY
[1335:1335:0208/074359.758356:ERROR:env.cc(257)] The platform failed to initialize.  Exiting.
```

I did `sudo init 3`, `sudo init 5` to switch runlevels but to no avail. `sudo reboot` solved it 😛

csabal, 2024-02-10:
Hi

/boot/check_for_httpd solution did not work.
There is a cursor. 

I am using the latest version from the mirror. (2023-12-11-fullpageos-bookworm-armhf-lite-0.14.0)

Also splash screen .png could not be loaded. 

Is there anybody who can tell me what version is actually working? Tired an old one from 2018 on a PI4 but it is just not works at all....

This is a very nice project, but now it is unfortunately  useless indeed.  :(

guysoft, 2024-02-26:
Hey, so there was an issue with the nighties which was fixed on the 2014-02-23. YOu can try it now.
Related #498

guysoft, 2024-06-10:
Closing if no reply

AlfaOmega777, 2024-06-12:
Hi, i'm also experimenting this behavior. Unfortunately i can't see all the boot process because i'm not physically with the device. But for example, executing the refresh bash, fixed it. I'm serving a react app using pm2 and serve command. The OS loads locally the webpage through localhost:3000. So any idea?

guysoft, 2024-06-13:
@AlfaOmega777 None that I can think of :-/

AlfaOmega777, 2024-06-13:
> @AlfaOmega777 None that I can think of :-/

Don't worry, this project is a great idea, but as a developer I can tell you just relax and try to make it ultimately simpler. For example, yesterday I took a standard bullseye 11 distro from raspberry pi imager. Almost only modifying the autostart file of the raspberry pi you can achieve the same functionality that fullpageos already has. I can tell you that all of us sometimes in our brain overthink a lot everything, when sometimes the key is make it as simple as possible.

guysoft, 2024-06-13:
@AlfaOmega777 FullpageOS has a lot of small things that were added to it over the years. It really depends what you wanna do with it. The fact that FullPageOS is the same image that has been run and tested by tens of thousands tends to be its strong point. But if you want to do something different, using your own could be your way.

stgreimel, 2024-06-20:
Hi!
I have the same problem!
https://orf.at - works
https://kultur-hafnerbach.at - black screen after boot section. I can connect via ssh but even sudo reboot does not work, I have to power off the device.

guysoft, 2024-06-30:
@stgreimel What version of FullPageOS?