guysoft/FullPageOS #511: Some emoji doesn't display properly

jkriegshauser, 2024-02-04:
#### What were you doing?

Certain emoji (in particular i noticed this with "☁️") does not display properly on the Chromium that comes with FullPageOS, though other emoji does.

#### What did you expect to happen?

The page as viewed from Chrome running on Windows:
[image]

Some emojis are fine:
[image]

#### What happened instead?

The page as viewed from Chromium on FullPageOS:
[image]

Note that the cloud emoji is not colored.

The house emoji (🏠) is fine:
[image]

#### Was there an error message displayed? What did it say?

No.

#### Version of FullPageOS?

[Can be found in /etc/fullpageos_version ALWAYS INCLUDE.]

0.13.0

#### Screenshot(s) showing the problem:

[If applicable. Always include if unsure or reporting UI issues.]

See above.

#### If you are building FullPageOS - provide a build.log that is created for the build

Downloaded stable image.

I also tried installing:
```
sudo apt install fonts-noto-color-emoji
```
and restarted, to no avail

jkriegshauser, 2024-02-04:
I was able to fix it by loosely following [this page](https://www.omgubuntu.co.uk/2016/08/enable-color-emoji-linux-google-chrome-noto), though it has some errors.

What worked:
Add the following to `~/.config/fontconfig/fonts.conf` (note that the page above is missing `<fontconfig>`)
```xml
<?xml version="1.0" encoding="UTF-8"?><!DOCTYPE fontconfig SYSTEM "fonts.dtd">
<fontconfig>
 <match>
  <test name="family"><string>sans-serif</string></test>
  <edit name="family" mode="prepend" binding="strong">
   <string>Noto Color Emoji</string>
  </edit>
 </match>

 <match>
  <test name="family"><string>serif</string></test>
  <edit name="family" mode="prepend" binding="strong">
   <string>Noto Color Emoji</string>
  </edit>
 </match>

 <match>
  <test name="family"><string>Apple Color Emoji</string></test>
  <edit name="family" mode="prepend" binding="strong">
   <string>Noto Color Emoji</string>
  </edit>
 </match>
</fontconfig>
```

Flush the font cache with:
```sh
fc-cache -f -v
```

Reboot or load the page and it shows the proper cloud emoji:
[image]

guysoft, 2024-02-04:
Hey, if you can post your full working ``~/.config/fontconfig/fonts.conf`` I could add a fix for this.

jkriegshauser, 2024-02-05:
It’s posted in full above

SecularSteve, 2024-02-07:
If I may make a suggestion, OpenMoji recently got a full update to the Unicode 15 standard:

https://github.com/hfg-gmuend/openmoji/releases/tag/15.0.0

I personally think it would be handy to implement it as the default Emoji setting that comes built-in with the system.
Could double as a fall-back in case of further errors like these.

guysoft, 2024-02-07:
@SecularSteve Thanks for the tip:
Related, this is what we do now: https://github.com/guysoft/FullPageOS/issues/378

guysoft, 2024-02-20:
Nightly build are working again.

@jkriegshauser If I make a build with a fix for it could you test it?

jkriegshauser, 2024-02-20:
@guysoft Sure

guysoft, 2024-03-06:
didn't forget, I want to fix the username bug first (https://github.com/guysoft/FullPageOS/issues/525)

guysoft, 2024-06-18:
Didn't forget, but I am really really swamped.