guysoft/FullPageOS #521: Symlink broken!

csabal, 2024-02-21:
Symbolic link does not work between the wpa config file, and the fullpageos config file!

It results that if you modify the fullpageos-wpa-supplicant.txt file it does not affect the wifi settings at all!
So, you got only black screen, because there is no net configured (either eth. or wifi)

Check:
ls -l /etc/wpa_supplicant

You should see this:

[image]

If not, run this:
sudo ln -sf /boot/fullpageos-wpa-supplicant.txt /etc/wpa_supplicant/wpa_supplicant.conf

And check again!

Reboot, and wifi settings applied.

guysoft, 2024-02-21:
There is an issue with the bookworm build, see #498
Working on it

guysoft, 2024-02-24:
See comment https://github.com/guysoft/FullPageOS/issues/498#issuecomment-1962347265
I am closing here, feel free to comment there. 
I also tested wpa supplicant network method setting and it worked.

closed 2024-02-24 by guysoft

csabal, 2024-04-11:
It is still not working!!! Not at all.

If you modify the fullpageos-wpa-supplicant.txt nothing happens, since it is not forward the settings to the wpa_supplicant.conf. 

Unfortunately the above command just don't work anymore.
Nightly built from the PI Imager.

Any help?????

Very sad.

csabal, 2024-04-11:
OK, so the weird thing it is forward the WIFI data to the wpa_supplicant/ wpa_supplicant.conf BUT the Pi connect to the original WIFI what was given during the install. 

I am confused...and give this up.

guysoft, 2024-04-11:
Are you using the release candidate from 2024-03-31?
i tested it. Flashing from rpi imager works

csabal, 2024-04-11:
Yes!

Just try to modify the fullpageos-wpa-supplicant.txt !!! It wont pick up other Wifi, still connecting that one, I entered durign the rpi imager!!!!