guysoft/FullPageOS #533: Raspberry Pi 5 - not connecting to wifi

JeromeTR2909, 2024-03-12:
#### What were you doing?

Used Raspberry Pi Imager v1.8.5 to install FullPageOS 2024-02-23. Install worked.
Did the custom config on the host name and wifi settings.

#### What did you expect to happen?

Go to the local webpage on my network.

#### What happened instead?

It boots up correctly but lands on a black screen with the cursor. On my Pi4, it works perfectly, loads and go to the black screen with cursor but after a few seconds it loads the page. On the Pi5, it stays on that forever.
It seems like it does not connect to the Wifi because when I check on my Unifi console, it does not show up as a client. But the Pi4 does. If I use an ethernet cable then it works. But wifi is a big no no. 

#### Was there an error message displayed? What did it say?

No error message, only black screen with cursor

#### Version of FullPageOS?

FullPageOS (Nightly) 2024-02-23 (even tried 2024-02-22)

[Can be found in /etc/fullpageos_version ALWAYS INCLUDE.]

#### Screenshot(s) showing the problem:

[If applicable. Always include if unsure or reporting UI issues.]

#### If you are building FullPageOS - provide a build.log that is created for the build

JeromeTR2909, 2024-03-15:
I believe I have solved this issue for me...

If you use the Raspberry Pi Imager to install FullPageOS 2024-02-23 from the option it has and set the wifi setting in the customize section, it will not work.

I had to download the img file from the repo - https://unofficialpi.org/Distros/FullPageOS/nightly/ (this is the one i downloaded: 2024-02-23_2023-12-11-fullpageos-bookworm-armhf-lite-0.14.0.zip)
and use the imager to install this img and then set your custom options. and afterwards do the config in the wpa file as well then it will connect to the wifi easily.

guysoft, 2024-04-05:
Hey, thanks for that information, will try and look in to what is the cause.

guysoft, 2024-04-07:
Hey, I just tested on 0.14.0 RC1 and it worked for me.
1. I opened it and navigated to the "Other specific use operiuting systrems" > fullpageos> build from 31.3.2024. and it.
2. Set the wifi SSID+Password + username+password

Pi 5 booted normally and has wifi, could ssh to it.

Did not edit any txt file.

Could it be you have some other complication? What wifi settings are you using?

csabal, 2024-04-07:
Hi,

I have used this image: 2024-03-15-fullpageos-bookworm-armhf-lite-0.14.0.img

1. Used the custom image file in the Raspberry Pi Imager
2. set the Wifi SSID + pwd, + username (Pi)+pwd. Simple 2,4Ghz wifi, wpa/wpa2. I also clicked the SSH option enabled in the Raspberry Pi Imager.  After first boot, took out the SD card, and I saw that no wifi information in the config file. It was untouched. Modified, and put back to the Pi4, and it found the wifi. The weird thing that the Pi username+pwd worked (I could ssh in) , so it seems the imager transferred that information, but not the Wifi stuff.  

It is Pi 4, btw.

Hope it helps.

csabal, 2024-04-07:
Hi.

@I am not sure that it is related to here, pls move it in any other thread if not. 

I have working in a non-profit museum , your SW could make my life easier.... :)

Feature requests:
1, Can you pls integrate this, as an option, during the install: (or maybe a new txt file in the boot partition)
https://github.com/guysoft/FullPageOS/wiki/Hide-mouse-cursor
2.  Could you pls also integrate this: (txt or any other method would be fine)
https://github.com/Linkek/keyboardTriggerServer 
This is the right keyboard layout: https://stackoverflow.com/questions/70574505/how-to-change-the-default-matchbox-keyboard-layout
FIle name: keyboard-lq1.xml 
Because these has all the international characters....worldwide.  SO, everybody will be fine using this layout!

These are absolute necessary, IF you use FullpageOS as it meant to be used.... :) For example in a museum, where NO cursor is a MUST, and also collecting visitor feedback, with full touch keyboard w international characters, using a nice KIOSK is also needed.

I have made the above, by hand, it works perfectly. 

Really appreciate your work! 

cs.

guysoft, 2024-04-07:
> Hi,
>
> I have used this image: 2024-03-15-fullpageos-bookworm-armhf-lite-0.14.0.img
>
>     1. Used the custom image file in the Raspberry Pi Imager
>
>     2. set the Wifi SSID + pwd, + username (Pi)+pwd. [...]

Hey I think I understood your issue.
1. Rpi imager. when you flash using custom does not apply the custom changes. The reason is that it does not know what type of image you are flashsing. They have in the yaml that publishes the image (that is not availbale in the custom). A setting that gives that information:
```
init_format	"systemd"
```

2. Please flash using the .xz image file, no need to extract it. The file name should have the date build.
the xz file name is  ``2024-03-31_2024-03-15-fullpageos-bookworm-armhf-lite-0.14.0.zip`` . Note the two dates:
``2024-03-31`` for the fullapgeos build date. And ``2024-03-15`` for the rpios build date.

closed 2024-04-07 by guysoft

guysoft, 2024-04-07:
Regarding your feature requests.
FullPageOS was not initially built to run only in a museum. I built it for dashbaords. And since then it was used in (to my knowledge): shops, factories, TV stations, Churches, companies that build photo booths and many more.

You are more than welcome to fork and build CustomPiOS with these changes. I have a full time job and no one pays me to develop specific features upon request. If there is a working variant I can even add it to the build pipeline of the current FullPageOS, but you have to maintain it. Also, if the current changes helped you in your job (which it seems so in the museum which one BTW? :) ) , there is a donation button in the README, this really helps me keep the project going and add hardware and make time fo rit.

VVD2501, 2025-03-26:
> I believe I have solved this issue for me...
>
> If you use the Raspberry Pi Imager to install FullPageOS 2024-02-23 from the option it has and set the wifi setting in the customize section, it will [...]

do you mean the wpa_supplicant.conf file? and it looks like the below?

country=CA
ctrl_interface=DIR=/var/run/wpa_supplicant GROUP=netdev
update_config=1

network={
    ssid="wifiname"
    psk="wifipassword"
}

VVD2501, 2025-03-26:
> It is still exist in 0.14.0 RC1. Workaround: let it boot the Pi after the FW flashed. Wait for the SSH generation and other stuffs, the Pi gonna reboot several times. Than switch off, take out the SD card, and modify the txt-s. After reboot it will be okay.

what is the txt-s file? can you please be more specific?

guysoft, 2025-03-27:
@VVD2501 its part of the readme:

https://github.com/guysoft/FullPageOS?tab=readme-ov-file#how-to-use-it

tl;dr - ``wifi.nmconnection``