guysoft/FullPageOS #534: Raspberry Pi 4- Multiple tabs not loading/booting

Argetlam3, 2024-03-16:
#### What were you doing?

I edited the fullpageos.txt file to have http://localhost/fullpagedashboard, as indicated by the setup page. I then went to the fullpagedashboard.txt file and put in the sites https://google.com and https://github.com (as test sites). When I booted up the pi with the micro sd card in it, nothing but my mouse and a black screen loaded. However, when I tried editing the fullpageos.txt file to only include one site, it booted successfully.

#### What did you expect to happen?

The sites to load and shuffle

#### What happened instead?

A blank screen with my mouse

#### Was there an error message displayed? What did it say?

N/A

#### Version of FullPageOS?

Can't seem to access, but it is the stable version on the raspberry pi imager.
#### Screenshot(s) showing the problem:

n/a
#### If you are building FullPageOS - provide a build.log that is created for the build
n/a

Vesa-Stremia, 2024-03-22:
Using the Nightly 2024-02-23 I get only the welcome page when trying to use the dashboard/multiple tabs.

What I've noticed while trying to troubleshoot, is that best way to get the FullPageOS working more consistently, is to burn the image -> run the OS once in your Rasp -> take the SD card back to your PC and now modify needed files. If I burn the image and right after modify the files, something goes haywire and I get to black screen or something else.

But still can't figure out how to get the multiple website rotation to show up.

guysoft, 2024-03-22:
Can you please provide your ``fullpagedashboard.txt``?

Vesa-Stremia, 2024-03-22:
Here's mine:

[fullpagedashboard.txt](https://github.com/guysoft/FullPageOS/files/14722314/fullpagedashboard.txt)

guysoft, 2024-04-10:
Not sure why that would happen, I can't reproduce it.
You can try the latest RC1, that one I tested and confirm it works.

curlerhaus, 2024-05-01:
I'm having a similar issue. The single URL works when I input into fullpageos.txt. When I put multiple pages in fullpagedashboard.txt and point fullpageos.txt to http://localhost/fullpagedashboard the Rpi stops loading on the FullPageOS splash screen. I'm having the same issue on a Rpi5 and 2b. 
[fullpagedashboard.txt](https://github.com/guysoft/FullPageOS/files/15181588/fullpagedashboard.txt)

[fullpageos.txt](https://github.com/guysoft/FullPageOS/files/15181589/fullpageos.txt)

guysoft, 2024-06-10:
@curlerhaus If its stuck on the slash screen it means that there is some connection issue. FullPageOS will not start the browser before it can connect to the url.
Does it work if you don't set the addresses?

curlerhaus, 2024-06-10:
If I leave fullpageos.txt blank I get in terminal:

 /usr/lib/chromium-browser/chromium-browser: no process found

When I input http://localhost/fullpagedashboard and leave fullpagedashboard.txt blank i get the same error.

curlerhaus, 2024-06-10:
If i leave fullpageos.txt blank I get the same result as pointing to http://localhost/fullpagedashboard. It just stays on the splash screen. 

When the fullpageos.txt file is set to http://localhost/fullpagedashboard it stays on the splash screen. Running the script reload_fullpageos_txt does not load the pages. When fullpageos.txt is set to blank and I try to run the reload script I get the following error:

/usr/lib/chromium-browser/chromium-browser: no process found

Not sure if this is anything but I'm using sudo nano to edit the files via ssh. When I type in a valid website the text turns blue. The http://localhost/fullpagedashboard typed in does not turn blue.

guysoft, 2024-06-18:
```
/usr/lib/chromium-browser/chromium-browser: no process found
```
Strange - could it be an sd card error? try switching sd cards?
Sounds like files that are part of the install are missing.

RDRGSJ00, 2024-08-29:
I was having a similar issue, when I add the URLs via the dashboard app (http://192.168.10.119/fulldashboard), it works but when I add them via the .txt files on the SD card it won't work. 
Info below:
- Pi: Raspberry Pi 4
- OS: FullpageOS (Nightly) 2024-03-31

apartmedia, 2024-08-29:
> I was having a similar issue, when I add the URLs via the dashboard app (http://192.168.10.119/fulldashboard), it works but when I add them via the .txt files on the SD card it won't work. Info below:
> 
> * Pi: Raspberry Pi 4
> * OS: FullpageOS (Nightly) 2024-03-31

See [here](https://github.com/guysoft/FullPageOS/issues/587#issuecomment-2294589799), that might be the reason and explain the functionality (and also one of the bugs and limitations when it comes to multiple site rotation in the current nightly).

FernandoLorenzi, 2024-10-04:
Hello friends, did you find a solution for two urls? Thanks.

apartmedia, 2024-10-04:
Did you read the previous post? It mentions another issue/comment with limitations. Note the info about iFrames in general. This is by design and can not be solved really by any chromium landing page. It would need a complete different approach and technique. You should first check if the websites you are planning to load in parallel and rotate are affected by this limitations. If yes, you can not use the current FPOS solution (or better say not the FP Dashboard solution).