guysoft/FullPageOS #537: chroium white page/Blank Page after rotating the screen

jeffreyx185, 2024-04-02:
Hey i use FullpageOs on different devices. In the New Version, after rotating the screen (like the config.txt settings in the FAQ) the Chromium Browser shows a white Page. Its after Disable the dtoverlay.

Same Error on Rasperry Pi Zero 2W, Raspberry 3, Raspberry 4.

Looks like something is not working with Chromium. I also test different Websitesfor example google.com 

Any Ideas how i can fix the Problem?

guysoft, 2024-04-02:
Which version is the "new" version? Do you mean the nightly build from 31st of Match 2024?

Also - can you provide the exact files of the settings you changed?

Does the default page work?

jeffreyx185, 2024-04-02:
Hi Thanks for Quick response.

Yes i used the latest nightly Build from 31st of March but after that i used the Stable Build from 7st of March 2022.

Both had the same Problem.

Here is what i did.

I Install the Img via the Raspi imager.

- then change the WPA Supplicant with my SSID and PSK.
First Boot: Eversthing work as intended.

Then i changed the /boot/config.txt to rotate the Screen.

# dtoverlay=vc4-kms-v3d
# increase gpu_mem for display rotate
gpu_mem=256
# Rotate HDMI screen
# 1 is 90 degrees. 2 is 180 degress. 3 is 270 degrees.
display_hdmi_rotate=1
# only for lcd screens # 180?
# display_lcd_rotate=1
# only used for syncing touch wich screen 1=180? flip?
# lcd_rotate=1

Like that. After the reboot the Boot Monitor is in the correct Orientation. But, when Chromium starts its not loading the Website. Also not the default page.
If i undo the Config, and put back the dtoverlay=vc4-kms-v3d everything works insteat of the rotation.

guysoft, 2024-04-14:
Hmm, if both had the same problem - it might be worth trying to flash the official raspberrypi desktop preform the same changes there, and see if the problem is there too.
If it isn't then there is a difference between them that we could try and pick up.
FullPageOS is based off RpiOS. If there are display bugs there, its the same driver and should behave the same.

Also, you can try using the non-acceleration variant, that helps sometimes.