guysoft/FullPageOS #538: Not work hotkeys on me page (chromium window not focused)

Vetal7979, 2024-04-02:
#### What were you doing?

Start my page in chromium, on page use hotkey

#### What did you expect to happen?
the keys will work

#### What happened instead?
the keys not work, because chromium window not focused (not active)
if i click mouse on window (anywhere on the screen) keys begin work correctly
#### Was there an error message displayed? What did it say?

#### Version of FullPageOS?

FullpageOS Nightly (2024-03-31)
in version Stable (2022-03-07) all works correctly

#### Screenshot(s) showing the problem:

[If applicable. Always include if unsure or reporting UI issues.]

#### If you are building FullPageOS - provide a build.log that is created for the build

guysoft, 2024-04-14:
Saw, will find a time to reproduce and debug

AlfaOmega777, 2024-04-20:
Same problem here, using Raspberry PI 3b+ and a barcode scanner.

guysoft, 2024-05-08:
I didn't forget this, but I am swamped, I was going to check this out as a possible solution: https://superuser.com/questions/142945/bash-command-to-focus-a-specific-window

pwoehrl, 2024-08-28:
Any possible updates? Will also look into the link 👍

guysoft, 2024-08-29:
@pwoehrl really don't have the time to implement this.

pwoehrl, 2024-08-29:
I will try 👍