guysoft/FullPageOS #542: Otot.tv outdated

Pebkac03, 2024-04-13:
When browsing to learn about fullpageos the discrepancy on otot.tv and GitHub confused me, especially on the subject of the experimental browser mentioned in the setup instructions on otot.tv.

From what I can gather the website hasn't been updated since 2019 and I am guessing the mailing list isn't active. If so, it would help if the website is removed or if a note is added that the site is deprecated since (insert date here).

I was also wondering if the things mentioned with regards to the planned custom browser are still on the roadmap.

guysoft, 2024-04-14:
@tailorvj Maintains that page

tailorvj, 2024-04-14:
@Pebkac03 Thanks for reaching out. There is actually a (slow) development underway for a new browser. 

I've noticed the recently growing interest in the website and the new browser and it gives me energy to keep working on this. I am also open for collaboration with JS and DevOps devs

If you're interested in it, please feel free to contact me. I'd love to know more about your use case and it will probably give me more energy to push new test features and eventually a release

@guysoft thanks for tagging me. I'll start looking into CustomPiOS integration

Kind regards,
Tailor
tailor.vj@gmail.com

Pebkac03, 2024-04-15:
Hi, thanks for the reply both of you! Just got a newsletter email as well. Might I suggest adding published and last edited datemarks to articles on the site.  So as to to make it easier to find out how current the information is.

It makes me glad to hear that the new browser is underway!

nomadicinteractiveinc, 2024-04-21:
I am too curious about this experimental browser.  I've signed up and happy to hear from you Tailor.