guysoft/FullPageOS #543: Website requires Google Sign in to access

tomisanhues2, 2024-04-17:
Hello,

I have an issue regarding my current use case for FullPageOS, I need to access a website which is protected by Authentication which required Google Authentication to login. This page depends heavily in JavaScript, the website is built in React, NextJS14 and TailwindCSS. Which the website updates itself via API every 5 seconds, so it's a "real time" site.

How can I get this use case to work correctly on FullPageOS, I leave some screenshots as follows:
[image]

First image button works perfectly with redirect

[image]

Pressing log in with google doesn't do anything, on any other browser, even chromium it works, even, pressing CTRL + "Sign in with google" opens a weird mode in FullPageOS which allows the entire process perfectly, but, after logging in this weird mode and correctly authentication, if I return closing the extra tabs, the site doesn't refresh meaning the authentication is not complete, it's almost like no JS is being executed in the main page.

guysoft, 2024-04-19:
FullPageOS is running chromium in kiosk mode. So its strange you are saying it does run in chromium.
Could you try pressing "ctrl+t" and using a browser windows that is not in kiosk mode and see if that helps?

bpeck, 2024-05-19:
Hello, I am running into a similar issue today. My intention is to display a tab in the multi tab dashboard that is logged into my google calendar.

Using the most recent nightly build available on the RPi Imager today, I made a fresh image and ran the OS on a RPI 4. When attempting to load the URL https://www.calendar.google.com into a tab added via the on-screen UI of the multi tab dashboard, I get a frowny face icon (the page does not load.)

If I hit ctrl+t to open a non kiosk window of chromium, I am able to login to a google account and open this URL without any issues.

I will try having Full Page OS configuration load the google calendar URL directly, instead of in the iframe dashboard and report back if that works.

Thanks