guysoft/FullPageOS #544: Can't get composite out for Fullscreen OS

lucypaw26, 2024-04-17:
#### What were you doing?
I'm using a pi 4b and I can't seem to get FullPage OS to output through the 3.5mm composite jack. I'm trying to run it to an old CRT TV with composite input. I'm using a breakout cable to RCA to plug in video and audio on the TV.

#### What did you expect to happen?
I edited the config.txt of the boot file in the SD card to include lines like:

hdmi_ignore_hotplug=1
enable_tvout=1
sdtv_mode=2
sdtv_aspect=1

I hoped this would bypass the HDMI output and go through the 3.5mm jack instead. I get some shaking/signal through my CRT on startup but then it quickly stops. I can plug in the HDMI cable and then the HDMI monitor screen loads with my webpage from the OS.

Any help is appreciated! I am very new and trying to learn.

#### Version of FullPageOS?

I am using the latest official release, FullPageOS 0.13.0

[Can be found in /etc/fullpageos_version ALWAYS INCLUDE.]

guysoft, 2024-04-19:
Does it work for you using plain Rpi OS?

chriskewl, 2024-12-22:
Latest Nightly (10-28-2024) this works fine on a Pi4, just use sudo raspi-config and set the composite that way and reboot. This can be closed. :)

guysoft, 2024-12-24:
Indeed

closed 2024-12-24 by guysoft