guysoft/FullPageOS #545: Unable to rotate touch screen

DanielLBarlow, 2024-04-21:
#### What were you doing?

Trying to rotate display

#### What did you expect to happen?
The display to rotate

#### What happened instead?
The display did not rotate

#### Was there an error message displayed? What did it say?
No

#### Version of FullPageOS?

0.14.0

#### Screenshot(s) showing the problem:

[If applicable. Always include if unsure or reporting UI issues.]

#### If you are building FullPageOS - provide a build.log that is created for the build

**More details**
I am running FullPageOS on a Raspberry Pi 3B to display the dashboard for my Home Assistant. I'm using the official 7 inch display. I have seen multiple threads online about adding lcd_rotate=2 and other similar things to config.txt. But this does not seem to be working for me.
I have tried to add it to the config in /boot and the config in /boot/firmware where the config in /boot tells me it is now. Makes no difference for me.
With there being so many different threads on the issue and none that seem to kind of collate it all, i was getting very confused on how i can solve this issue.
So any help that can be provided for this would be greatly appreciated

DanielLBarlow, 2024-04-21:
So i have noticed now that when i do lcd_rotate=2. During the reboot process, it starts off in the orientation i want, but then flips back again a few seconds after when it's loading everything through

frutabruta, 2024-04-28:
’’’
sudo nano /home/pi/scripts/start-gui
’’’
And there set
 ’’’
DISPLAY_ORIENTATION=left
’’’

preacher4gh, 2024-07-30:
Same situation here.
Raspberry Pi3 with official 7" touschscreen, where rotation by 180 degree is needed.
I tried a lot of settings in cmdline.txt (cold rotate boot messages), config.txt (nothing worked) and start_gui.
With some settings I could rotate the second logo appearance, but never the website to show. The webpage did not rotate in any setting.
When setting orientation in "start:gui" from "nornal" to "inverted", the reaction of the mouse-cursor on touch input flipped, but not the picture.

Thins I tried in config.txt:
uncomment dtoverlay=vc4-kms-v3d : black screen
video=DSI-1:800x480M@60,rotate=180 : black screen
lcd_rotate=2, display_lcd_rotate=2  :  no change

cmdline.txtx:
add fbcon=rotate:2 : orientation of boot messages correct

switch to wayland in raspi-config: black screen
error loading X when swithing back to X11, stuck at login-promt

Did someone already find a solution?

apartmedia, 2024-07-31:
Using of lcd_rotate is deprecated and doesn't work with the DRM VC4 V3D Driver (dtoverlay=vc4-kms-v3d)

What you can try (tested with FullPageOS 0.14.0 and the official RPi 7“ Touchscreen)

**Rotate the screen in X upside down (180 degrees)**
`xrandr —orientation inverted` (should work and FullPageOS is using this in start_gui if you set DISPLAY_ORIENTATION=inverted)
what also worked for me:
`xrandr —output DSI-1 —rotate inverted`

**Rotate the touch input accordingly**
As far as I could see this is currently not automatically handled by FullPageOS, so doing it manually is necessary.
Open libinput.conf (in /usr/share/X11/xorg.conf.d/40-libinput.conf).
Find the touchscreen section…
`Section „InputClass“`
`     Identifier „libinput touchscreen catchall“`
and set the corresponding 180 degree rotation angle by adding a "CalibrationMatrix“ option…
`     Option "CalibrationMatrix" "-1 0 1 0 -1 1 0 0 1"`

Reboot the Pi an you should have a 180 degree rotated screen and touch.

Adding to cmdline.txt (not config.txt)
`video=DSI-1:800x480M@60,rotate=180`
is another way of fully rotating the DSI screen and also flips the fbi splash image, but unfortunately in my tests this messed up chromium and the website view and was therefore unusable.
As screen rotation Rotating also needs rotating the touch the Foundation recommends configuring this in the input library or desktop and not via device tree entries, as this can cause conflicts with your input library.

Adding to cmdline.txt (not config.txt)
`fbcon=rotate:2`
will rotate the console (and purely the console/boot messages, which I want to suppress completely in kiosk mode anyway).
Any other framebuffer application will remain at 'normal' orientation and needs to be configured separately.

Most of the regular documentation refers to configuring this with Wayfire, the Wayland desktop compositor which is the new standard in Raspberry Pi OS bookworm. FullPageOS seems to use the X11 window system which was used in the previous OS versions (e.g. buster, bullseye).

Hope this helps a bit.

tsull360, 2025-05-22:
> Using of lcd_rotate is deprecated and doesn't work with the DRM VC4 V3D Driver (dtoverlay=vc4-kms-v3d)
>
> What you can try (tested with FullPageOS 0.14.0 and the official RPi 7“ Touchscreen)
>
> **Rotate [...]

What would the value for 90 degrees of touch input rotation be?