guysoft/FullPageOS #546: Trying to get a Rapsberry Pi 3 Model B to Rotate 90 degrees Fullpage OS on a Raspberry PI 7" Touchpanel

JackBebbington, 2024-04-23:
#### What were you doing?
Fullpage OS is working with a website and touch functionality works.
I would like the website to be 90 degrees so I opened /bootfs/config.txt
at the bottom of the page I added: lcd_rotate=2

[Please be as specific as possible here]

#### What did you expect to happen?
7" Touchpanel to show the page rotated in portrait mode at 90 degrees.

#### What happened instead?
7" Touchpanel did not rotate the image at all.

#### Was there an error message displayed? What did it say?

#### Version of FullPageOS?
0.13.0

guysoft, 2024-06-10:
Try this?
https://github.com/guysoft/FullPageOS/issues/545#issuecomment-2081254684

JackBebbington, 2024-06-17:
Thanks guysoft! This worked for the display, however the touchpanel is still out on landscape calibration

guysoft, 2024-06-18:
Hmm, you can try reverting what you did and then changing this:
https://forums.raspberrypi.com/viewtopic.php?p=1734808&sid=e600b2000b2822a15b4d3804a022cc47#p1734808

I haven't tried it myself