guysoft/FullPageOS #554: Timeout / Lockout

Colebiagetti, 2024-04-30:
### Discussed in https://github.com/guysoft/FullPageOS/discussions/551

<div type='discussions-op-text'>

<sup>Originally posted by **Colebiagetti** April 26, 2024</sup>
Hello, currently using fullpage OS to display a browser on boot. I find that it is timing out after 10 minutes. Is there a specific file/folder in the OS that I can change it so that it never times out?</div>

guysoft, 2024-04-30:
Hey, the screen should remain on with FullPageOS by default. I am closing this issue and please continue talk in the discussion.

guysoft, 2024-05-30:
What version of FullPageOS? Can you try version 0.14.0 nightly?

georgygoshin, 2024-06-17:
I made an image with raspberry imager and I too have black screen after 10 minutes. Where to switch it off?

novo-github, 2024-09-04:
> @georgygoshin Same question to you - what version of FullPageOS? 0.14.0 has changes to avoid a blank screen.

I had this bug in 0.13.0, and nightly / 0.14.0 fixed it. Thank you

wlans, 2024-11-02:
> > @georgygoshin Same question to you - what version of FullPageOS? 0.14.0 has changes to avoid a blank screen.
> 
> I had this bug in 0.13.0, and nightly / 0.14.0 fixed it. Thank you

I had to set up only wifi in the imager for nightly build 14 from October 24th

numbworks, 2025-04-15:
Thank you guys for this thread!

I installed **FullPageOS 0.13.0** (_2022-03-07_2022-01-28-fullpageos-bullseye-armhf-lite-0.13.0.zip_) on my **Raspberry PI 3**, but the screen timeout issue was very annoying. 

Googling around I found this thread and I switched to **FullPageOS 0.14.0** ("_2025-04-01_2024-11-19-fullpageos-bookworm-armhf-lite-0.14.0.zip_"). Now the timeout issue is gone. 

In any event, configuring the WiFi via **Raspberry Pi Imager**'s options didn't work for me, therefore I share the notes I wrote for myself in the case somebody else will end up in this thread and need some help! These notes apply to **Windows 10**.

**SETUP FULLPAGEOS 0.14.0:**

1. Plug a microSD into the computer;
2. Download and install **RasperryPi Imager**;
3. Download the "_2025-04-01_2024-11-19-fullpageos-bookworm-armhf-lite-0.14.0.zip_" from here:
   https://unofficialpi.org/Distros/FullPageOS/nightly/
4. Burn the ISO image on the microSD;
5. Eject the microSD from your computer and reinsert it to show "_/bootfs_";
6. Open the "_/bootfs_" folder and edit/save the following files:

	- _/bootfs/fullpageos.txt_: replace "_http://localhost/fullpagedashboard_" with your URL of choice
	- _/bootfs/wifi.nmconnection_: open and add the content of the local "_wifi.nmconnection.txt_"

7. Eject the microSD from your computer and insert it into the **Raspberry Pi**;
8. Done!

**SETUP AUTOREFRESH ON FULLPAGEOS**:

1. Install **TightVNC**;
2. Open **TightVNC** and connect to "_fullpageos.local_";
3. Press CTRL+T to exit from Chrome's fullscreen and open the **Chrome Web Store**;
4. Install the "**Easy Auto-Refresh**" extension and configure it for your URL(s);
5. Done!

**Side Note**: for me _fullpagedashboard_ doesn't work, neither in 0.13.0 nor in 0.14.0.