guysoft/FullPageOS #556: My FullscreenOS SetUp has an active Screensaver - how to disable?

Baggerfahrer, 2024-05-02:
#### What were you doing?
I setup a fresh new installation of FullPageOS via rasperry pi imager this week on my raspberry pi 4 with an attached HDMI touchscreen. 

#### What did you expect to happen?
According to the FAQ of FullPageOS, there is no screensaver or switch off of the Screen after a specific amount of time; but my installation is going to idle after +- 5min. See attached video.
Whenever i touch the screen, the system is waked up again.

#### What happened instead?

#### Was there an error message displayed? What did it say?
No

#### Version of FullPageOS?
fullpageos 0.13.0

[Can be found in /etc/fullpageos_version ALWAYS INCLUDE.]

#### Screenshot(s) showing the problem:
https://streamable.com/ea71r4

I think important for this issue is the content of 'start_gui' file in scripts folder. I did not change this content, but for documentation reasons i pasted here the content.

`#!/bin/bash

# Stop the screen from turning off when idle.
DISABLE_POWER_MANAGEMENT=yes

# Rotate screen if needed, see 'xrandr -h' for options.
DISPLAY_ORIENTATION=normal

if ["${DISPLAY_ORIENTATION}" != 'normal'];
then
  xrandr --orientation ${DISPLAY_ORIENTATION}
fi

if ["${DISABLE_POWER_MANAGEMENT}" == 'yes'];
then
  xset s off         # don't activate screensaver
  xset -dpms         # disable DPMS (Energy Star) features.
  xset s noblank     # don't blank the video device
fi

compton -b -d :0 --backend glx --unredir-if-possible --glx-swap-method buffer-age --glx-no-stencil --paint-on-overlay --vsync opengl 
matchbox-window-manager &

/home/pi/scripts/run_onepageos

`

I would kindly ask for support to switch off the screensaver function.
Many thanks!

Baggerfahrer, 2024-05-05:
no one an idea?

guysoft, 2024-05-05:
Duplicate of #496
Closing here so we can continue the talk in a single place

guysoft, 2024-05-05:
A quick google search says it might be a powersave feature on the screen:
https://github.com/raspberrypi/linux/issues/1351

closed 2024-05-05 by guysoft