guysoft/FullPageOS #561: DISPLAY_ORIENTATION Not working on Bullseye Stable Build

salpoosi, 2024-05-15:
RPI 4B
Bullseye Stable Build 0.13.0

I'd like to use screen rotation, but it didn't work. I found an issue in the file, so I'm providing the correct content for it to function properly. Please apply the changes.

/home/pi/scripts/start_gui

-if ["${DISPLAY_ORIENTATION}" != 'normal'];
+if [ "${DISPLAY_ORIENTATION}" != 'normal' ];

-if ["${DISABLE_POWER_MANAGEMENT}" ==  #'yes'];
+if [ "${DISABLE_POWER_MANAGEMENT}" == 'yes' ];

closed 2024-05-27 by guysoft

JonathanPorta, 2024-05-28:
I wanted to chime in, mainly for anyone else having this issue, and mention that as of 5/23/24, when using the Raspberry Pi Imager, the stable version of FullPageOS available with that tool was built on 2022-02-07 and does not yet contain the bug fix @guysoft mentioned. [1]

@guysoft Also, thanks a ton for this project. It's great!

[1]: https://github.com/guysoft/CustomPiOS/commit/8bbe19d0ce87c4969207663e5f2ea682f6c895ae

salpoosi, 2024-05-29:
I think FullPageOS is a very good project and has features that are not at all inferior to other commercial versions. While I am aware that the source has already been fixed in CustomPiOS, the bug still persists in the official Raspberry Pi distribution of FullPageOS, and I would have liked to see them reflected in the official distribution image.
Thank you for this project. @guysoft

guysoft, 2024-06-02:
@JonathanPorta @salpoosi Thanks for the kind words. It really helps when there are so many bugs and issues to get control over to hear it :)

If you can confirm that the issue is fixed in 0.14.0 RC1 which has an issue tracking it here:  #540
I would very much want to release it. It looks like the wpasupplicant issue there is something general and this means that its likely good enough for release.

nordurljosahvida, 2024-06-23:
@salpoosi @guysoft @JonathanPorta I'm very sorry to tag you guys, I've bee trying to work this issue for the better part of three days and my brain is on fire.

Raspi 4. Original 7" screen.

Boots as landscape, need to rotate view and of course touch interface too.

I was using stable version on raspi, display_rotate / lcd_rotate didn't work. Disabling dtoverlay made the screen black.

I luckily found this post, so I downloaded 0.14.0 RC1, flashed it now.

What I don't understand is, once I flash this version, which of the 37000 ways of rotating the screen I need to use.

Is it by running [this script](https://github.com/guysoft/CustomPiOS/blob/devel/src/modules/gui/filesystem/opt/scripts/rotate.sh) from the booted os?

Thank you so much in advance!

guysoft, 2024-06-30:
Hey, there are several ways to do this. I don't have a running instance to test it for you though.

Maxinoha1, 2024-09-18:
Can I ask for advice on how to set the rotation by 90 degrees - I tried it:
DISPLAY_ORIENTATION=right
but after reboot the image did not rotate.
I also tried to update the entire content of the file /home/pi/scripts/start_gui, see the above error.

guysoft, 2024-09-23:
@Maxinoha1 Did you try it with the nightly? Version 0.13.0 I think has an issue with it.

Maxinoha1, 2024-09-24:
> @Maxinoha1 Did you try it with the nightly? Version 0.13.0 I think has an issue with it.

I tried the Stable version (2022-03-07) which is available in the Raspberry Pi Imager app.

I tried the Nightly version once, but the Raspberry Pi Zero kept restarting.

I would like to run a Raspberry Pi Zero W or Zero 2W with a Wisecoco 5.5" OLED display.

I'll try the Nightly version again.

Maxinoha1, 2024-09-24:
[image]

> @Maxinoha1 Did you try it with the nightly? Version 0.13.0 I think has an issue with it.

I tried the Stable version (2022-03-07) which is available in the Raspberry Pi Imager app.

I tried the Nightly version once, but the Raspberry Pi Zero kept restarting.

I would like to run a Raspberry Pi Zero W or Zero 2W with a Wisecoco 5.5" OLED display.

I'll try the Nightly version again.

Maxinoha1, 2024-09-25:
I was unable to get the Nightly version up and running. Raspberry Pi Zero 2 W did not connect to WiFi.
So I tried the older version 0.12.0 and used the command to rotate the display:
`DISPLAY=:0 xrandr --output HDMI-1 --rotate right`

guysoft, 2024-11-12:
@Maxinoha1 The wifi configuration in the nightly is different since raspberry dropped support for wpa supplicant and moved to NetworkManager https://github.com/raspberrypi/bookworm-feedback/issues/72 .

You need to use now the [wifi.nmconnection](https://github.com/guysoft/CustomPiOS/blob/devel/src/modules/network/filesystem/network-manager/boot/wifi.nmconnection) file.