guysoft/FullPageOS #562: no boot erro on screen

maxpaine69, 2024-05-17:
#### What were you doing?
installing FullPageOs on a pi zero 2w.

[Please be as specific as possible here]

#### What did you expect to happen?
boot the Os and show something, but not an error

#### What happened instead?

its showin a error :

#### Was there an error message displayed? What did it say?

"Xsession: unable to lauch "/home/scripts//start_gui" X session --- "/home/pl/scripts/start_gui" not found falling back to default session."

#### Version of FullPageOS?
0.13.0

[Can be found in /etc/fullpageos_version ALWAYS INCLUDE.]

#### Screenshot(s) showing the problem:

[image]

[If applicable. Always include if unsure or reporting UI issues.]

#### If you are building FullPageOS - provide a build.log that is created for the build

guysoft, 2024-05-30:
It says there is a file missing - perhaps the SD card file is damaged? Can you reflash on a different card and try again?

Seriousness, 2024-06-02:
Did you create a custom user during flashing the sd? I've had a custom user name and had to manually create pi plus copy the scripts folder

guysoft, 2024-06-04:
@Seriousness That is fixed and tested in the current release candidate

SirMaple, 2024-11-02:
FYI. This issue is still present. I created a custom user name in the Raspberry Pi Imager and got the same error. If I change the username in the Imager to pi the issue is gone during boot.

Running on a Raspberry Pi 3

guysoft, 2024-11-02:
@SirMaple what version of FullPageOS did you flash?
Do you get the issue if you set user to Pi?

lmfreitas, 2024-12-29:
> @SirMaple what version of FullPageOS did you flash? Do you get the issue if you set user to Pi?

That's it, in raspberry pi imager the setting default user prevents image from having user "pi" and all files in "/home/pi/" are ignored.

Remove default user and everything works as expected.