guysoft/FullPageOS #567: Embedded video playback

IT-Mase, 2024-05-29:
#### What were you doing?

Displaying a webpage

#### What did you expect to happen?

In the center of my page there's an mp4 video file playing automatically (via an elementor video widget).
Was working on standard raspbian/chromium

#### What happened instead?

Video is not playing and an Chromecast logo appears in place of the video

#### Was there an error message displayed? What did it say?
No
#### Version of FullPageOS?

0.14.0

#### Screenshot(s) showing the problem:

[image]

guysoft, 2024-05-29:
Do you have the page? Does it work on regular raspberrypi stock image?

SlyLaM, 2024-05-30:
Hi,
yes it work on a regular raspbian image with chromium.
I made a page for testing purposes:
https://tinyurl.com/2325ek93

Also to note that your chromium has no standard google fonts installed, so many sites do not display the way they should.

Thank You

guysoft, 2024-05-30:
No idea there, perhaps there is a plugin missing?

SlyLaM, 2024-05-30:
Maybe, but why do I need to install anything on your veersion ?

guysoft, 2024-06-03:
Raspbian (now rpi os) ship the distro with pre-installed plugins. I am not sure how they install them to be honest, chromium did not include a way to install extension from command-line list time I checked. 
FullPageOS installed the GUI on top of a Rpi OS lite, which does not include a UI. There might be differences.