guysoft/FullPageOS #568: Grafana not loading, possible old chromium version?

EHGTube, 2024-06-03:
#### What were you doing?

Updated Grafana over the Weekend to its newest Version ( atleast thats what my logs tell me). 
Now Chromium fails to load the application files. 
Access to Grafana from a different PC works so im guessing its fullpageos. 

The Full Error Message is: 

`If you're seeing this Grafana has failed to load its application files

1. This could be caused by your reverse proxy settings.

2. If you host grafana under subpath make sure your grafana.ini root_url setting includes subpath. If not using a reverse proxy make sure to set serve_from_sub_path to true.

3. If you have a local dev build make sure you build frontend using: yarn start, or yarn build

4. Sometimes restarting grafana-server can help

5. Check if you are using a non-supported browser. For more information, refer to the list of supported browsers.`

1-3 are not applicable.
4: I tried restarting Grafana and the Raspberry but it didnt help.
So I am left with point 5. 

(but I have no Idea if thats actually the issue, I am just guessing that the Chromium I am using is old)
Running Chromium-browser --version gives: `Chromium 88.0.4324.187 Built on Raspbian , running on Raspbian 10`

Is there anything I can do to update the Version of fullpageos I am using? 
(like apt-get update?)

guysoft, 2024-06-03:
Chromium is FullPageOS 0.12.0 which was release in 2021. You can try upgrading to FullPageOS 0.13.0 which has Chromium Version 98.0.4758.106  . Or even to the nightly build which is also the release candidate for 0.14.0 which has Chromium Version 122.0.6261.89

closed 2024-06-03 by guysoft

EHGTube, 2024-06-03:
okay cheers. Is there any way of updating the os without reflashing the sd card ?
like apt-get update/upgrade?

EHGTube, 2024-06-03:
Okay cheers. 
Took a bit to get everything working again but it works. 
Thank you very much for the OS, its running my "Status" TV, showing some vital information for my services and has proven very useful so far!