guysoft/FullPageOS #569: FullPageOS is staying on the splash screen.

cenktrms, 2024-06-03:
#### What were you doing?

Boot

[Please be as specific as possible here]

#### What did you expect to happen?

Open my page and show.

#### What happened instead?

Stayed on FullPageOS boot screen with mouse cursor.

#### Was there an error message displayed? What did it say?

No error message

#### Version of FullPageOS?
0.14.0

[Can be found in /etc/fullpageos_version ALWAYS INCLUDE.]

#### Screenshot(s) showing the problem:

[image]

[If applicable. Always include if unsure or reporting UI issues.]

#### If you are building FullPageOS - provide a build.log that is created for the build

guysoft, 2024-06-05:
If you see the splash screen and a cursor it means that the GUI has loadad, its the background of the desktop. And the browser is not switching on because its waiting for an internet connection

closed 2024-06-05 by guysoft

guysoft, 2024-07-24:
@rafael-monkeysoft 
security.org is not a registered domain to you, I assume the error is that you don't have a signed certificate for https, also its strange doing https on port 8085

carloavilam-gif, 2025-08-27:
i tried doing so but it could not find the path.
it will sometimes log half the page. half of the pictures will be unable to load

guysoft, 2025-08-28:
@carloavilam-gif That sounds like a connectivity issue. Check your dmesg for connection or other IO errors