guysoft/FullPageOS #570: WiFi not working / SSH on LAN not enabled

mortomanos, 2024-06-04:
#### What were you doing?

Flashed the recent version of FullPageOS 0.14.0 on an SD card, configured the 7" display correctly, configured WiFi correctly

#### What did you expect to happen?

WiFi should connect and should get the DHCP registration, SSH login via IP should work

#### What happened instead?

WiFi did not work (no IP address assigned), so I tried to boot with LAN connected, got an IP address, but SSH was not enabled on this IP address

#### Was there an error message displayed? What did it say?

no error message

#### Version of FullPageOS?

Flash from 0.14.0: https://unofficialpi.org/Distros/FullPageOS/nightly/2024-03-31_2024-03-15-fullpageos-bookworm-armhf-lite-0.14.0.zip

#### Screenshot(s) showing the problem:

n/a

#### If you are building FullPageOS - provide a build.log that is created for the build

guysoft, 2024-06-04:
How are you setting up wireless?

Note that due to a change in RpiOS wpa_supplicant autoconfigure 
I am not sure what we can do about it
https://github.com/raspberrypi/bookworm-feedback/issues/72

guysoft, 2024-06-04:
This might be a solution:
https://github.com/raspberrypi/bookworm-feedback/issues/72#issuecomment-1848778447

mortomanos, 2024-06-04:
The problem is that if I cannot login to the shell, I cannot do anything on the filesystem...

mortomanos, 2024-06-06:
I forgot that there might be a local console, so I attached a keyboard and switched to console 1 and was able to login. I noticed that no sshd was running. I then saw that there is a sshswitch service which looks for /boot/firmware/ssh (which was not present). I cannot say for sure why this did not work either after creating /boot/firmware/ssh, but I had to enable ssh manually - now at least I can logon from remote on LAN and can continue to try getting WiFi working.

mortomanos, 2024-06-06:
I can confirm that the solution presented in the comment above (adding a configuration file for NetworkManager) does work for me. Maybe it would make sense to just do it the same way, have a configuration file in /boot and copy it in the destination (and change the access rights to 0600 which is important).

guysoft, 2024-06-06:
That file should be created in the build process here:
https://github.com/guysoft/CustomPiOS/blob/devel/src/modules/base/start_chroot_script#L120

I can also see that in 0.14.0 RC1 the line did execute like this
```
 + touch /boot/firmware/ssh
 ```
 
 Also, if I open the image on my computer (using the [chroot_in_to_image.sh](https://github.com/guysoft/CustomPiOS/blob/devel/src/chroot_in_to_image.sh) script), I see the ssh file in place
 
 ```
 $ ls -l  mount/boot/firmware/ssh 
-rwxr-xr-x 1 root root 0 Mar 31 12:37 mount/boot/firmware/ssh
```

wlans, 2024-11-02:
### Step 1: Flash FullPageOS and Configure Wi-Fi

1. **Download FullPageOS**:
   - Download the FullPageOS image file, such as `2024-10-28_2024-10-22-fullpageos-bookworm-armhf-lite-0.14.0.zip`, to your computer.

2. **Open Raspberry Pi Imager**:
   - Install and open [Raspberry Pi Imager](https://www.raspberrypi.org/software/).

3. **Select OS and Storage**:
   - **Choose OS**: Click “CHOOSE OS,” select “Use Custom,” and choose the FullPageOS image file.
   - **Choose Storage**: Click “CHOOSE STORAGE” and select your microSD card.

4. **Configure Wi-Fi in Advanced Options**:
DO NOT SET A SSH USER OR ANY USER
   - Press `Ctrl + Shift + X` (or `Cmd + Shift + X` on macOS) to open the Advanced Options.
   - **Enable Wi-Fi**: Check “Configure wireless LAN.”
   - **Enter Network Details**: Input your Wi-Fi **SSID**, **password**, and **Wi-Fi country**.

5. **Write the OS to the microSD Card**:
   - Click “WRITE” to flash FullPageOS with Wi-Fi settings. Eject the microSD card when complete.

6. **Insert the microSD Card and Boot**:
   - Insert the card into the Raspberry Pi, connect it to a monitor and keyboard, and power it on. The Pi should automatically connect to Wi-Fi.