guysoft/FullPageOS #571: x11vnc needs to be changed to be user-independent

guysoft, 2024-06-05:
See here, has user pi:

https://github.com/guysoft/FullPageOS/blob/devel/src/modules/fullpageos/filesystem/root_init/etc/systemd/system/x11vnc.service#L7

Needs to be changed to something global

romankysely, 2024-09-14:
I agree. I have great experience with realVNC server instead  preinstalled x11VNC. I suggest to chang it at image. x11VNC is very slow nad demanding sources I think. 
RealVNC is following system/user password.
Or maybe consider to  preinstall Pi Connect service which is no require static public IP at all. But I do not have practical experience with it.

guysoft, 2024-09-22:
@romankysely I have no time to implement that, but would accept PRs if you have something that works.

commit 7a8f4385 (2024-11-03): VNC support for not user pi Fixes #571
closed 2024-11-03 by guysoft