guysoft/FullPageOS #573: Unable to connect to wifi

bezrq, 2024-06-10:
#### What were you doing?

Open webpage after installing through Pi Imager. I tried with and without preconfigured SSID, tried the wpa applicant file and imager config

#### What did you expect to happen?
Open and load webpage and connect to wifi
#### What happened instead?
Stuck on splash
#### Was there an error message displayed? What did it say?
No
#### Version of FullPageOS?

http://unofficialpi.org/Distros/FullPageOS/nightly/2024-03-31_2024-03-15-fullpageos-bookworm-armhf-lite-0.14.0.zip

[Can be found in /etc/fullpageos_version ALWAYS INCLUDE.]

#### Screenshot(s) showing the problem:
it’s just the dashboard. not connecting to wifi, no ip. 
[If applicable. Always include if unsure or reporting UI issues.]

#### If you are building FullPageOS - provide a build.log that is created for the build

bezrq, 2024-06-10:
using a zero 2 w

bezrq, 2024-06-11:
update: I am able to get chromium to work/load once I have booted and it goes to a white screen; ssh into pi and run 
`killall chromium-browser`

after around 5-10 sec it reloads chromium and goes to the page specified. im not sure why this is required for it to load, and i dont want to have to do this for it to work.

wlans, 2024-11-02:
Got it working by setting up only wifi in the imager for nightly build 14 from October 24th

wlans, 2024-11-02:
### Step 1: Flash FullPageOS and Configure Wi-Fi

1. **Download FullPageOS**:
   - Download the FullPageOS image file, such as `2024-10-28_2024-10-22-fullpageos-bookworm-armhf-lite-0.14.0.zip`, to your computer.

2. **Open Raspberry Pi Imager**:
   - Install and open [Raspberry Pi Imager](https://www.raspberrypi.org/software/).

3. **Select OS and Storage**:
   - **Choose OS**: Click “CHOOSE OS,” select “Use Custom,” and choose the FullPageOS image file.
   - **Choose Storage**: Click “CHOOSE STORAGE” and select your microSD card.

4. **Configure Wi-Fi in Advanced Options**:
DO NOT SET A SSH USER OR ANY USER
   - Press `Ctrl + Shift + X` (or `Cmd + Shift + X` on macOS) to open the Advanced Options.
   - **Enable Wi-Fi**: Check “Configure wireless LAN.”
   - **Enter Network Details**: Input your Wi-Fi **SSID**, **password**, and **Wi-Fi country**.

5. **Write the OS to the microSD Card**:
   - Click “WRITE” to flash FullPageOS with Wi-Fi settings. Eject the microSD card when complete.

6. **Insert the microSD Card and Boot**:
   - Insert the card into the Raspberry Pi, connect it to a monitor and keyboard, and power it on. The Pi should automatically connect to Wi-Fi.

guysoft, 2024-11-02:
@wlans Are you an AI bot? why are you spamming with irrelevant messages bunch of issues?