guysoft/FullPageOS #578: White dots on the bottom corners

bitcoinlimit, 2024-06-23:
Hey guys,

Hope I can get some answers here. I'm building a smart frame dashboard using FullPageOS but I see two white dots on the bottom left and one white dot on the bottom right whenever FullPageOs is active in fullscreen mode.

These white dots are never visible when I view the dashboard on other browsers/computers either in browser mode or fullscreen mode. Unfortunately my dashboard's background is black and these annoying dots are visible because they're also flickering.

I've tried both Nightly releases (03.24 and 03.31) but the result is the same.

I'm on Raspberry Pi 5.

I suspect this might be an issue related to the hidden mouse cursor but again I'm not sure because there are multiple flickering white dots on different corners of the bottom.

Please let me know if anyone had a similar issue before or how I could fix it.

Appreciate your helps.

Thank you,

guysoft, 2024-06-30:
Can you take a photograph so I can see how the dots look like?

bitcoinlimit, 2024-06-30:
Here is a very short video I recorded. They are too small. two on the left bottom and one on the right bottom.

https://github.com/guysoft/FullPageOS/assets/173568805/9560c3a1-3b24-4ff3-9819-f5d91dd0cd2f

Thanks for looking into this.

bitcoinlimit, 2024-07-12:
Any update on this?

Thanks!

guysoft, 2024-07-13:
Hey, I am not sure what is causing it, it might be a bug in chromium or the window manager. I don't have time to debug it - how adventurous are you to debug this yourself?
You could try switch to firefox, or change the window manager.