guysoft/FullPageOS #579: [feature request] Auto releoad when tab crashes (aw Snap message appears)

Bluscream, 2024-06-30:
#### What were you doing?

Trying to run a 24/7 dashboard for my HomeAssistant instance

#### What did you expect to happen?

It just working infinitely

#### What happened instead?

Crash after some time

#### Was there an error message displayed? What did it say?

Aw, snap!

#### Version of FullPageOS?

blu@dashboard:~ $ cat /etc/fullpageos_version
0.14.0

#### Screenshot(s) showing the problem:

[image]

### What i already tried:

- Setting up "official" autorefresh: https://github.com/guysoft/FullPageOS/issues/127#issuecomment-1134599760
- Setting up autorefresh from https://github.com/thomasloven/hass-browser_mod/blob/master/documentation/services.md#browser_modrefresh via a HomeAssistant Automation
- asking AI for help but im too dumb to write addons https://www.phind.com/search?cache=z89rtzecmrx7gk7wtzruizhq

guysoft, 2024-08-06:
There is a system for that.

The code is here:
https://github.com/guysoft/FullPageOS/blob/devel/src/modules/fullpageos/filesystem/opt/custompios/scripts/start_chromium_browser#L27

It looks for the words "ERROR" or "FATAL" and refreshes. Can you provide logs of a crash so we can see why it didn't work?

Bluscream, 2024-08-06:
I got around this by adding a SSH integration for the dashboard into homeassistant and just issuing ssh reboot commands every 3 hours. I'll get back to this if im ever bothered by that

closed 2024-08-06 by Bluscream