guysoft/FullPageOS #580: After boot flashes FPOS then black screen with cursor flashing irregularly

CrashClark, 2024-07-01:
First thanks and great job helping bring this forward. I'm a NOOB, just dangerous enough to break my pi's when I try new things.

#### What were you doing? just rebooting after trying to add pi user to www-group with this cmd 'sudo usermod -a -G www-data pi' an example I found on the interwebs. I was having issues getting FPOS to display the localhost webpage that was being served out by Apache. it only showed the smaller FPOS graphic for some reason.

#### What did you expect to happen? load a local php webpage with image rotator using Apache2

#### What happened instead? during boot, lines of text loading down the page, clear screen some more lines of text, flash FPOS image then flashes a black screen with a cursor top left corner that blinks some code sequence of long and short flashes it seems very irregular after awhile it does just short flashes. the webpage is actually being served out over my network and can be view by other PCs on the net. 

#### Was there an error message displayed? What did it say? no error,

#### Version of FullPageOS? 0.14.0 dtd 3/15/24

guysoft, 2024-07-01:
Strange, FullPageOS 0.14.0 should show the logo when it waits for internet and then loads the browser. So this would mean the GUI is not loading at all. Can you get the output of ``/var/log/Xorg.0.log``? 
Another thing might be that the monitor and FullPageOS are not working well together. Did it work when you just flashed and booted a plain FullPageOS flash without any changes?

Also, there is already apache2 installed I don't think you need to add the pi to the www-data group.

CrashClark, 2024-07-01:
It was showing both the large FPOS image and then the smaller version during boot at the same time that its IP was serving out the correct PHP page that it should have been showing.
do you know how I can reverse the usermod I did? I guess I can just start over again and see where it goes off the rails.

log:

[    55.325]
X.Org X Server 1.21.1.7
X Protocol Version 11, Revision 0
[    55.325] Current Operating System: Linux NewOSNMENU 6.6.35-v8+ #1779 SMP PREEMPT Tue Jun 25 11:37:38 BST 2024 aarch64
[    55.325] Kernel command line: coherent_pool=1M 8250.nr_uarts=0 snd_bcm2835.enable_headphones=0 snd_bcm2835.enable_headphones=1 snd_bcm2835.enable_hdmi=1 snd_bcm2835.enable_hdmi=0  smsc95xx.macaddr=DC:A6:32:BD:1A:3F vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  console=ttyS0,115200 console=tty1 root=PARTUUID=ec752406-02 rootfstype=ext4 fsck.repair=yes rootwait logo.nologo consoleblank=0 loglevel=0 cfg80211.ieee80211_regdom=US
[    55.326] xorg-server 2:21.1.7-3+rpt2+deb12u7 (https://www.debian.org/support)
[    55.326] Current version of pixman: 0.42.2
[    55.326]    Before reporting problems, check http://wiki.x.org
        to make sure that you have the latest version.
[    55.326] Markers: (--) probed, (**) from config file, (==) default setting,
        (++) from command line, (!!) notice, (II) informational,
        (WW) warning, (EE) error, (NI) not implemented, (??) unknown.
[    55.326] (==) Log file: "/var/log/Xorg.0.log", Time: Mon Jul  1 01:32:31 2024
[    55.326] (==) Using config directory: "/etc/X11/xorg.conf.d"
[    55.326] (==) Using system config directory "/usr/share/X11/xorg.conf.d"
[    55.327] (==) No Layout section.  Using the first Screen section.
[    55.327] (==) No screen section available. Using defaults.
[    55.327] (**) |-->Screen "Default Screen Section" (0)
[    55.327] (**) |   |-->Monitor "<default monitor>"
[    55.327] (==) No monitor specified for screen "Default Screen Section".
        Using a default monitor configuration.
[    55.327] (**) Option "Debug" "dmabuf_capable"
[    55.327] (==) Automatically adding devices
[    55.327] (==) Automatically enabling devices
[    55.327] (==) Automatically adding GPU devices
[    55.327] (==) Automatically binding GPU devices
[    55.327] (==) Max clients allowed: 256, resource mask: 0x1fffff
[    55.327] (WW) The directory "/usr/share/fonts/X11/cyrillic" does not exist.
[    55.327]    Entry deleted from font path.
[    55.327] (==) FontPath set to:
        /usr/share/fonts/X11/misc,
        /usr/share/fonts/X11/100dpi/:unscaled,
        /usr/share/fonts/X11/75dpi/:unscaled,
        /usr/share/fonts/X11/Type1,
        /usr/share/fonts/X11/100dpi,
        /usr/share/fonts/X11/75dpi,
        built-ins
[    55.327] (==) ModulePath set to "/usr/lib/xorg/modules"
[    55.327] (II) The server relies on udev to provide the list of input devices.
        If no devices become available, reconfigure udev or disable AutoAddDevices.
[    55.327] (II) Loader magic: 0x20a080
[    55.327] (II) Module ABI versions:
[    55.327]    X.Org ANSI C Emulation: 0.4
[    55.327]    X.Org Video Driver: 25.2
[    55.327]    X.Org XInput driver : 24.4
[    55.327]    X.Org Server Extension : 10.0
[    55.330] (++) using VT number 7

[    55.330] (II) systemd-logind: logind integration requires -keeptty and -keeptty was not provided, disabling logind integration
[    55.332] (II) xfree86: Adding drm device (/dev/dri/card1)
[    55.332] (II) Platform probe for /sys/devices/platform/gpu/drm/card1
[    55.348] (II) xfree86: Adding drm device (/dev/dri/card0)
[    55.348] (II) Platform probe for /sys/devices/platform/v3dbus/fec00000.v3d/drm/card0
[    55.348] (**) OutputClass "vc4" setting /dev/dri/card1 as PrimaryGPU
[    55.348] (II) LoadModule: "glx"
[    55.349] (II) Loading /usr/lib/xorg/modules/extensions/libglx.so
[    55.351] (II) Module glx: vendor="X.Org Foundation"
[    55.351]    compiled for 1.21.1.7, module version = 1.0.0
[    55.351]    ABI class: X.Org Server Extension, version 10.0
[    55.351] (II) Applying OutputClass "vc4" to /dev/dri/card1
[    55.351]    loading driver: modesetting
[    55.351] (==) Matched modesetting as autoconfigured driver 0
[    55.351] (==) Matched fbdev as autoconfigured driver 1
[    55.351] (==) Assigned the driver to the xf86ConfigLayout
[    55.351] (II) LoadModule: "modesetting"
[... 8 lines cut ...]
[    55.353]    compiled for 1.21.1.3, module version = 0.5.0
[    55.353]    Module class: X.Org Video Driver
[    55.353]    ABI class: X.Org Video Driver, version 25.2
[    55.353] (II) modesetting: Driver for Modesetting Kernel Drivers: kms
[    55.353] (II) FBDEV: driver for framebuffer: fbdev
[    55.364] (II) modeset(0): using drv /dev/dri/card1
[    55.364] (WW) Falling back to old probe method for fbdev
[    55.364] (II) Loading sub module "fbdevhw"
[    55.364] (II) LoadModule: "fbdevhw"
[    55.364] (II) Loading /usr/lib/xorg/modules/libfbdevhw.so
[    55.364] (II) Module fbdevhw: vendor="X.Org Foundation"
[    55.364]    compiled for 1.21.1.7, module version = 0.0.2
[    55.364]    ABI class: X.Org Video Driver, version 25.2
[    55.364] (II) modeset(0): Creating default Display subsection in Screen section
        "Default Screen Section" for depth/fbbpp 24/32
[    55.364] (==) modeset(0): Depth 24, (==) framebuffer bpp 32
[    55.364] (II) Applying OutputClass "vc4" options to /dev/dri/card1
[    55.364] (==) modeset(0): RGB weight 888
[    55.364] (==) modeset(0): Default visual is TrueColor
[    55.364] (II) Loading sub module "glamoregl"
[    55.364] (II) LoadModule: "glamoregl"
[    55.365] (II) Loading /usr/lib/xorg/modules/libglamoregl.so
[    55.374] (II) Module glamoregl: vendor="X.Org Foundation"
[    55.374]    compiled for 1.21.1.7, module version = 1.0.1
[    55.374]    ABI class: X.Org ANSI C Emulation, version 0.4

guysoft, 2024-07-01:
Looks ok
Did it work when you just flashed and booted a plain FullPageOS flash without any changes?

CrashClark, 2024-07-01:
I really didn't just test that stage, but after installing Apache and updates/upgrades it seemed fine ie "FPOS" image was on, then I started configuring my website, as I said the website is serving it out, but after I tried to add pi to www-data with that example command above it turned to black. Not sure what else to try, do you know how i can reverse the 'sudo usermod -a -G www-data pi' I did?  as of now
$ groups =
'pi adm dialout cdrom sudo audio www-data video plugdev games users input render netdev gpio i2c spi'
thanks again for your support.

guysoft, 2024-07-01:
It really should not effect it that way, that is why I am not sure how to reverse it.
I think this is the reverse command:
```
sudo gpasswd -d pi www-data
```

CrashClark, 2024-07-01:
I'll give that try, I'm guessing I need to just start over and see if it works this time. I appreciate your time and suggestions.

nub1604, 2024-07-11:
Hi, I have the same strange behavior on my fresh installation from RPI Imager nightly build 2024-03-31 on a rpi3

The only thing I changed is the address in the fullpageos.txt
It seems the gui dont come up. 

[image]

guysoft, 2024-07-13:
@nub1604 What did you change it to?

nub1604, 2024-07-16:
@guysoft It was a link to a selfhosted simple blazer app. But it doesn't matter, I installed a fresh untouched installation of the 2024-03-31 build on a rpi3 and the results are the same. Only flickering prompt.