guysoft/FullPageOS #585: nightly builds are broken, stable version is broken, no working distribution

artbild, 2024-07-25:
Plain install from Raspberry Pi Imager. (Both nightly builds, 2024-03-07, 2024-03-31 and the stable version 2022 are broken 
tested on Raspberry Pi 4 and Raspberry Pi Zero 2W

#### Stable version (2022)
Stable version, after install is missing /home/pi/scripts/start_gui  (directories and file)  If I add it manually the hdmi output is broken and config.txt commands are ignored. Fullpageos.txt is ignored also but I can add the url in the start_gui script after I register the service. However the hdmi output is broken and ignores config.txt commands. I did not continue to override more things.

#### Nightly build 2024-03-31
Both current nightly builds are broken. The last nightly build 2024-03-31 does successfully accept config.txt commands but ignores fullpageos.txt url. Instead it loads a blank screen (users reported this on other forums and on youtube as well) Turning the display works, toggeling overscan works, specifying screen resolution also working fine. No connection is made to the link provided in fullpageos.txt

#### Nightly build 2024-03-07
This nightly build does ignore config.txt commands and also loads a blank screen. netstat proofes that there is no connection made to the link provided in fullpageos.txt

frutabruta, 2024-07-25:
The last working version for me is 2024_03_24, also installed from RPi Imager. Accepts webpage set in fullpageos.txt. I didn't need to edit config.txt, so that might not work. The only needed change was rotation of the screen - done via screen orientation parameter in start_gui script.

artbild, 2024-07-25:
Wich Pi did you run it on?

frutabruta, 2024-07-25:
Pi4, 2G RAM

artbild, 2024-07-25:
It does not work for me out of the box. I will try to stick with 03-31 and try to override fullpageos.txt since it does get ignored here. I did fiddle around with HDMI modes but that does not resolve the problem. However at least the config.txt commands do work fine here.

It's either a chromium problem or the fullpageos.txt is simply not loaded here.

frutabruta, 2024-07-25:
In one of the versions I remember that the system booted, but after I changed the orientation to "left" and rebooted, it got stuck on black screen.

artbild, 2024-07-25:
Okay, i read about the black screen problem and how it is fixed but that is not a problen here. The splash screen and the fullpageos picture are loaded fine, have the correct orientation and resolution. That really works fine here. For me it's just the url that does not load.

frutabruta, 2024-07-25:
And the SSH is working fine? Maybe it cannot connect to the internet. I guess you know that you can skip to the terminal via Ctrl+Alt+F1 to do the ’ip a’ or ’ping 8.8.8.8’ as a connection test.

artbild, 2024-07-25:
Yes the Pi can successfully sude apt update and I do all configuration with SSH. It is the distreibution that ignores fullapgeos.txt

guysoft, 2024-07-26:
1.
Hey @artbild i tested the latest nightly and it worked for me-  how are you flashing the image? Are you changing anything in the rpi-imager.
please provide reproduction stepss, otherwise I cant determine why it doesnt work for you, but does for me.
2. What forums? If you dont give a refence  have no idea what you are talking about

Slos, 2024-07-30:
I wonder who broke this? This project is dead (i.e. someone stabbed it to death). It was the greatest thing ever. Unbelievable.

guysoft, 2024-08-05:
@Slos glad to hear - did something not work for you before?

guysoft, 2024-08-06:
Can this be closed? @artbild any input as original poster?

Slos, 2024-08-08:
> @Slos glad to hear - did something not work for you before?

Hi,

Yes, some time ago I wanted to set up a kiosk using a raspberry Pi so I turned to FullPageOS. I tried configuring FullPageOS using the txt files contained in the image (fullpageos.txt, full-age-wpa-suplicant.txt, etc...). This did not work unfortunatelly (I opened an issue about that some time ago).

Now, last week I decided to give it another go but I used a different approach. I use the default fullpageOS dashboard instead of the txt system files to add my tabs/websites that I want to see on screen and everything works as expected. 

I also managed to customize the start_chromium_browser file to add a default zoom level to chrome when it starts up, this also worked fine.

One thing I did not manage to get working is the fullpageos-wpa-supplicant file so that after reboot the raspberry pi would connect to another wifi network than the one which was set up using the raspberry pi imager.

apartmedia, 2024-08-08:
> One thing I did not manage to get working is the fullpageos-wpa-supplicant file so that after reboot the raspberry pi would connect to another wifi network than the one which was set up using the raspberry pi imager.

As @guysoft already mentioned [in this issue here](https://github.com/guysoft/FullPageOS/issues/570) RPi OS bookworm dropped wpa_supplicant.conf support for configuring WiFi and moved to Network Manager. Therefore, the method used in FullPageOS to configure WiFi will currently not work, as it is no longer supported by Bookworm (and might never be in the future).
@guysoft is thinking about or already working on a replacement mechanism for the old wpa_supplicant.conf method.

Slos, 2024-08-08:
> As @guysoft already mentioned [in this issue here](https://github.com/guysoft/FullPageOS/issues/570) RPi OS bookworm dropped wpa_supplicant.conf support for configuring WiFi and moved to Network [...]

I see. Thanks for the info.

guysoft, 2024-08-08:
I am working on having a wpa-supplicant OR network manager file you can edit based on what the distribution has, but its moving slow. Its the last thing before I make a new RC

apartmedia, 2024-08-08:
> Its the last thing before I make a new RC

Will your new RC also address the bug regarding using multiple urls in fullpagedashboard.txt?
As far as I can see the FullPageDashboard code will not read multiple lines initially defined in fullpagedashboard.txt.
Therefore it’s currently not possible to configure multiple tabs and/or tab rotating by filling fullpagedashboard.txt with more than one line/one url.

jonnylink, 2024-10-22:
This seems completely broken. Using Pi 3 and it's missing all the startup files. Anything we can do to help?

guysoft, 2024-10-22:
@jonnylink What version of FullPageOS. what files are you missing?

Schnubl, 2024-10-24:
can confirm that.

i tryed now 4 images an no one is working, can see the splash screen that ins in a bootloop and after 3 times i get a message: from xsession that a file is missing can remember the name start_gui or something like that.

i moved on to an other product so i cant tell you more, but i didnt get it to work with 4 different versions Stable, Nightly, Nightly and one i found with non acceleration from 2022)

Always the same problems.

guysoft, 2024-10-27:
Hey,
@Schnubl and @artbild
I just tested 2024-03-31 on a Rpi 4, i flashed it from rpi imager, without any changes, plugged in an HDMI and ethernet cable it it booted to GUI fine. So I am not sure what is different with yours.

I am not sure what is different from what you have. I hope to get this merged, then build a new FullPageOS and release, my only guess is that you have Pi 4 that are different from mine and need a different firmware that was release in rpi os 2024-07-04](https://downloads.raspberrypi.org/raspios_lite_armhf/release_notes.txt). I started another nightly build so you can test that theory.

guysoft, 2024-10-28:
There is a new nightly build (2024-10-28). I tested it on my Pi and it boots fine. You are welcome to try it, its based on 2024-10-22 which came out today. It might solve your issues.

wlans, 2024-11-02:
> Stable version, after install is missing /home/pi/scripts/start_gui

Stable version, after install is missing /home/pi/scripts/start_gui  <-- you get this when you change the "config" in Raspberry PI Imager. I got it to.  If anyone is getting that clear your config in the imager.

phazoxNL, 2025-02-01:
3 different pi's from rpi4/5 and zero 2 same problems, wont even start properly.
no matter what you adjust. it is just a crap distro to be honest.