guysoft/FullPageOS #587: Only whitescreen in Chromium / no wifi

nrace87, 2024-08-16:
#### What were you doing?

Installed FullPageOs latest nightly build via Raspberry PI installer on a Raspberry Pi 2.
Added Wifi Information and 3 different URLs in fullpagedashboard.txt

#### What did you expect to happen?
Wifi was not working, but I could solve it via raspi-config quite easily. -> ok

Website is not displayed. Please confirm the following stettings are correct:

/boot/firmware/fullpageos.txt contains this entrance only:
http://localhost/FullPageDashboard

/boot/firmware/fullpagedashborad.txt contains multiple websites like:
http://localhost/welcome
http://www.google.com

With this settings I should see the welcomepage and the google websites alternating?

#### What happened instead?
Nothing, whitescreen in Chromium after FullpageOs Logo.

#### Was there an error message displayed? What did it say?
Just a blank Chromium page

#### Version of FullPageOS?

0.14.0

#### Screenshot(s) showing the problem:

[If applicable. Always include if unsure or reporting UI issues.]

#### If you are building FullPageOS - provide a build.log that is created for the build
N/A

Did I do something completly wrong? Tried with different image files.. once I had the welcome page displayed. 
It there somewhere an explanation how to setup the pages after the first boot? I read somewhere on github that the .txt file with the URLs is only loaded at "first boot"?

I read something about a GUI for configuration. I guess it is /scripts/start_gui? If I try to start it, chromium is just blinking the white screen but not showing anything.

Thanks for any help/ideas.

nrace87, 2024-08-16:
[image]
Just discoverd that.. looks a bit strange as the http is displayed at the end if I open the URL of the raspberry remotely.

apartmedia, 2024-08-16:
Does it work, if you delete the second URL entry in /boot/firmware/fullpagedashborad.txt?
Just use the one standard entry http://localhost/welcome (and only this in one single line).
Is the default welcome page loading?

nrace87, 2024-08-16:
Thats seems to be one of my problems here, I did not find any information how to reload the /boot/firmware/fullpagesdashboard.txt.

I changed entries there and tried to reload it with the scripts located in the scripts folder but seems not to work like this. Even restarting the whole raspberry does not reload the fullpagesdashboard.txt

Do I need to delete something that it can be re-read? I found the URLs in the json file as well, seems it does not change? Feel a bit like reengineering as I have no clue right now where to search :)

Thanks a lot for more tips and tricks.

apartmedia, 2024-08-17:
FullPageOS uses [FullPageDashboard](https://github.com/amitdar/FullPageDashboard) as the default App (located in the Webserver directory of FullPageOS) to display single or multiple websites. FullPageDashboard initially reads the web addresses from fullpagedashboard.txt. It then saves user defined sites in a file called urls.json in the web root dir. If this file exists, it will no longer read any entries from fullpagedashborad.txt in favor of urls.json. You can force reading from fullpagedashboard.txt again by deleting the urls.json file and reload the App.

Unfortunately there is a bug in the interaction between FullPageOS and FullPageDashboard.
FullPageDashboard can not successfully parse multiple init URLs in the way FullPageOS is presenting them in fullpagedashboard.txt (one separate line for each address).

Another possible problem could be, that not every website can be correctly displayed with FullPageDashboard as it uses iFrames to load each site content which is not supported by every website.

The easiest way if you just want to display a single page is placing that address into FullPageOS /boot/firmware/fullpageos.txt.
This will skip loading FullPageDashboard and just display this single website directly in chrome which should work for every address.

If you need multiple auto rotating websites based on FullPageDashboard functionality (and if they work using iFrames) you can try manually adding the addresses via the web GUI available on the standard FullPageDashboard welcome page (use the + at the upper address tab to manually add websites). This will disable fullpagedashboard.txt and create a urls.json file and instruct FullPageDashboard to read from there.
You can also try manually editing urls.json to specify your addresses (and also tab names) as you like (but you should take care of the correct json syntax).

Hope this helps…

nrace87, 2024-08-17:
Thanks a lot here a my latest results :)

I really like the deletion urls.json and reload, that works great.

Unfortunately I need that feature with mulitple websites. So I tried the workaround with the GUI (now I know what is meant by GUI for Fullpagesos). Unfortunately it does not work as well.

I noticed that it writes a urls.json (btw with a total different syntax than the one created with the initial fullpagesdashboard.txt) but it does not display any website, really tried multiple from google to my router logon page, wheater pages...

The only page here which seems to work is the https://localhost/welcome

Seems there is another problem here as well.. do you have any external website which should work 100% with this setup?
Thanks a lot

apartmedia, 2024-08-17:
[urls.json](https://github.com/user-attachments/files/16644276/urls.json)
You can try this test urls.json.
It rotates the standard welcome page with Wikipedia and Bootstrap. At least it displays something.
You will see that most websites prevent itself from embedding in iFrames. So in most cases this feature will not be very helpful unless you will use it for your own or local web content.
And again, keep in mind switching to FullPageOS 0.14.0 is the better start point. There are some bugs already fixed in this newer version. You can also install the new version from within rpi imager. Just select the latest nightly build.

nrace87, 2024-08-18:
Thank you very much, I am now some steps further. It really seems that a lot of websites do not support iFrames. 
The one I could more or less solve with a screenshot of the website I want to display.

I already use FullPageOS 0.14.0.

Is it possible if I use this gui mode to go to fullscreen? So that the single tabs are not longer visible?

Are there any ideas how to use websites with login credentials with fullpagesos or personal calenders like ical or directly the google calender?

Thanks :)

apartmedia, 2024-08-18:
> Is it possible if I use this gui mode to go to fullscreen? So that the single tabs are not longer visible?
>
There are ways to disable the FullPageDashboard top navigation bar.
Did you try this hint in the FAQ? [Can I hide the navigation bar on the top?](https://github.com/guysoft/FullPageOS/wiki/FAQ#Can-I-hide-the-navigation-bar-on-the-top)

> Are there any ideas how to use websites with login credentials with fullpagesos or personal calenders like ical or directly the google calender?
>
One possibility is using a regular mouse and keyboard to interact with the website in the standard way.
If you use a touchscreen it might be a solution install a virtual keyboard.
You can try the recommended idea from the FAQ….
[I want a virtual keyboard](https://github.com/guysoft/FullPageOS/wiki/FAQ#i-want-a-virtual-keyboard)

Romeo1984, 2024-08-21:
I just loaded the nightly build for the first time, I have an Ethernet connection. No modification of anything except the Pi Imager unique login information. I also have the "White Screen of Death" after boot and after package updating.

guysoft, 2024-08-21:
You are supposed to see the logo of FullPageOS untill the browser loads.
FullPageOS checks there is an internet connection before starting the browser.
If the browser started but its white, id try to plug a keyboard, press ctrl+t and you get a new windows you can try an interact with, put a known webpage, see if that works.

Romeo1984, 2024-08-21:
> You are supposed to see the logo of FullPageOS untill the browser loads. FullPageOS checks there is an internet connection before starting the browser. If the browser started but its white, id try [...]

Okay. This is weird. Today I powered up the test Pi to do what you asked, and it worked! I then tried the other builds that I flashed that was failing yesterday and they too all worked. I'm not sure what the heck happened yesterday...

arpieb, 2024-08-29:
Just flashed 0.14 nightly, only update was manually editing `fullpageos-wpa-supplicant.txt` to enable my WPA2 wifi creds + set the correct country code at the bottom.  I'm getting the FPOS logo, placeholder page loads up, but when I try to browse to a page within Chromium (using Ctrl+T) I'm getting "No internet."  Confirmed the MAC isn't showing up in my DNS table either.

Interestingly I had the opposite problem with the image installed by RPi  Imager (never did figure out which version) - wifi worked like a charm, but I could never get the window manager to launch, never saw a splash screen or dashboard page.

Irony is, I'm trying to set up a product launch countdown display for the team - and now I've wasted 3h trying to get this to work. 🤦

apartmedia, 2024-08-29:
AFAIK the stable version installed by RPi Imager is 0.13.0 which I believe is based on Debian 11 (Bullseye).
This version supports setting up WiFi with wpa-supplicant mechanism used by FPOS. But 0.13.0 has some other known scripting bugs preventing all documented features from working out if the box ([see here](https://github.com/guysoft/FullPageOS/discussions/524)).

If I understand the maintainer correctly, he has no plans to add updates to the old 0.13.0.
So switching to the latest nightly (0.14.0) is the way to go.
0.14.0 is based on the current stable Debian 12 (Bookworm).
Unfortunately RPi Foundation dropped support for the wpa-supplicant way for configuring WiFi and moved to Network Manager ([see here](https://github.com/guysoft/CustomPiOS/issues/219)).
So currently adding fullpageos-wpa-supplicant.txt to FPOS does not configure WiFi.
There already exists a PR (see [guysoft/CustomPiOS#227](https://github.com/guysoft/CustomPiOS/pull/227)) from the maintainer [guysoft](https://github.com/guysoft) to make this feature work again (using wpa-supplicant or Network Manager based on the installed OS version). I haven‘t been able to test it yet but at least there is something out to come and fix this.
If you rely on WiFi you currently need to configure it manually in FPOS 0.14.0 (you can do this with the build in raspi-config configuration tool) or you use a cable connection. Ethernet is working out of the box in the latest FPOS nightly.

Hope this helps a bit…

guysoft, 2024-08-29:
> Just flashed 0.14 nightly, only update was manually editing fullpageos-wpa-supplicant.txt to enable my WPA2 wifi creds + set the correct country code at the bottom. I'm getting the FPOS logo, [...]

@arpieb wpa-supplicant support has been dropped by raspberypi os: https://github.com/raspberrypi/bookworm-feedback/issues/72

I am in the process of adding a workaround with an option to set up a connection via NetworkManager, its what is blocking me from releasing FullPageOS 0.14.0, progress is shown here:
https://github.com/guysoft/CustomPiOS/issues/219

Pr for that is here:
https://github.com/guysoft/CustomPiOS/pull/227

SRC-RIT, 2024-11-07:
Wifi with WPA and wired network is working fine for me!

But a have the issue with :
/boot/firmware/fullpageos.txt contains this entrance only:
http://localhost/FullPageDashboard

/boot/firmware/fullpagedashborad.txt contains multiple websites like:
http://localhost/welcome
http://www.google.com/
(Tried alot of alternativs)

This does not work for me, tried deleting urls.json and it did not do the trick!
loading one page from fullpageos.txt works just fine!
But the rotating feature of multiple websites would be nice!

apartmedia, 2024-11-07:
See [here](https://github.com/guysoft/FullPageOS/issues/587#issuecomment-2294589799) for more details and possible reasons. It explains the functionality and also one of the bugs and limitations when it comes to multiple site rotation in the current versions.
Note the info about iFrames in general. This is by design and can not be solved really by any chromium landing page. It would need a complete different approach and technique. You should first check if the websites you are planning to load in parallel and rotate are affected by this limitations. If yes, you can not use the current FPOS solution (or better say not the FP Dashboard solution).

apartmedia, 2024-11-07:
> /boot/firmware/fullpagedashborad.txt contains multiple websites like: http://localhost/welcome http://www.google.com/ (Tried alot of alternativs)
>
> This does not work for me, tried deleting urls.json [...]

Duplicate of #534

Gun-neR, 2025-09-01:
I installed 0.14.0 on RPi 3B+   Everything seems to be working fin with the exception of a solid minute of white screen, after the **FullScreenOS Logo** (which itself tasks about 20 seconds) before it loads my webpage.  Is this normal?

I setup my webpage in `/boot/firmware/fullpageos.txt` as I had read somewhere here.