guysoft/FullPageOS #589: Boots to white screen even using default URL

prcyuk, 2024-09-10:
#### What were you doing?

Booting the Pi using both default local URL and my own webpage.

#### What did you expect to happen?

A webpage to load.
I changed the URL in both dashboard.txt and fullpage.os separately and restored to default, but still no webpage loaded.

#### What happened instead?

Just a white screen with the cursor in the lower right corner. I can move the cursor and using ctrl + t creates a new tab and I can access any webpage with the address bar, it just won't launch from boot.
I have looked through existing issues but their issues suggest that the default local URL works fine, whereas mine doesn't.

#### Was there an error message displayed? What did it say?

No message. Just cursor.

#### Version of FullPageOS?

0.13.0

Raspberry Pi Zero 2 W
Hyperpixel 4 Rectangle

closed 2024-09-10 by prcyuk

guysoft, 2024-09-11:
Hey @prcyuk Hope you managed to fix it?

prcyuk, 2024-09-11:
Changing from https to http seemed to allow it to boot to webpage so I
closed the issue for the time being 🙂

guysoft, 2024-09-11:
Its likely because the website you are trying to view has enabled "blocked Cross-Origin". A website can tell the browser "do not load me in an iframe". Its a security thing. If you don't use FullPageDashboard, it should load, but you wont have the app that lets you load multiple tabs.