guysoft/FullPageOS #591: Support RPi5

CFenner, 2024-09-18:
#### What were you doing?

I want to use FullpageOS on a RPi5. So I installed FullpageOS 2022-03-07 via Raspberry Pi Images 1.8.5.

#### What did you expect to happen?

#### What happened instead?

The RPi starts stating "The installed os does not indicate support for Raspberry Pi 5. Update the OS.

#### Was there an error message displayed? What did it say?

#### Version of FullPageOS?

FullpageOS 2022-03-07

[Can be found in /etc/fullpageos_version ALWAYS INCLUDE.]

#### Screenshot(s) showing the problem:

[If applicable. Always include if unsure or reporting UI issues.]

#### If you are building FullPageOS - provide a build.log that is created for the build

CFenner, 2024-09-18:
I also loaded https://github.com/guysoft/FullPageOS/releases/tag/0.13.0 and tried that, but no luck.

apartmedia, 2024-09-18:
Did you try the nightly version from RPi Imager?
The stable is version 0.13.0 based on bullseye with some issues in FPOS.
The nightly is version 0.14.0 with several bugfixes and based on the latest Debian bookworm RPiOS.
In general that should also work on RPi 5 (that being said I didn’t try by myself as I don‘t have a RPi5).
Just give it a try…

CCFenner, 2024-09-18:
Thanks, now I tried the 2024-03-24 from RPi Imager, that worked!

CFenner, 2024-09-18:
Thanks, now I tried the 2024-03-24 from RPi Imager, that worked!

guysoft, 2024-09-18:
Related:
https://github.com/guysoft/OctoPi/issues/823

The nightly should work fine on 5, the reason I am not releasing yet is because of:
https://github.com/guysoft/CustomPiOS/issues/219

chriskewl, 2024-12-22:
I have been installing this with the latest Nightly (10-28-24) on a Pi5 without issue, even with the customizing options through the Pi Imager. This should be good now and can close.

guysoft, 2025-01-02:
I want to close it after release because I need to update the json so it shows up as supported

pwbriggs, 2025-04-06:
@guysoft is this updated yet?

Stable still doesn't seem to be working on my Pi 5, but the 1 Apr 2025 Nightly boots fine.

guysoft, 2025-04-07:
@pwbriggs stable is not expected to work. I think I will have to release a new RC2 over this one, it looks like it would work better for most:
https://github.com/guysoft/FullPageOS/issues/540