guysoft/FullPageOS #594: Portrait Orientation

conrad9900, 2024-10-02:
#### What were you doing?

Trying to set screen orientation to portrait

#### What did you expect to happen?

To reboot and see portrait orientation

#### What happened instead?

No change

#### Was there an error message displayed? What did it say?

No Change

#### Version of FullPageOS?

0.14.0

[Can be found in /etc/fullpageos_version ALWAYS INCLUDE.]

#### Screenshot(s) showing the problem:

[If applicable. Always include if unsure or reporting UI issues.]

#### If you are building FullPageOS - provide a build.log that is created for the build

apartmedia, 2024-10-02:
You are not explaining in detail what exactly you did already to change the screen orientation, what display you want to rotate and if you also need to rotate the (touch) input.
You can take a look at this [issue](https://github.com/guysoft/FullPageOS/issues/545) to get some hints. This [comment](https://github.com/guysoft/FullPageOS/issues/545#issuecomment-2260941912) shows some foundation recommendations regarding screen and touch rotation with the official 7“ RPi DSI Touchscreen. It might be helpful in general when it comes to setting up screen orientation for any display.

guysoft, 2024-10-15:
Closing if no answer form OP. Thanks @apartmedia for the reply.

wlans, 2024-11-02:
### Step 1: Flash FullPageOS and Configure Wi-Fi

1. **Download FullPageOS**:
   - Download `2024-10-28_2024-10-22-fullpageos-bookworm-armhf-lite-0.14.0.zip` to your computer.

2. **Open Raspberry Pi Imager**:
   - Install and open [Raspberry Pi Imager](https://www.raspberrypi.org/software/).

3. **Select OS and Storage**:
   - **Choose OS**: Click “CHOOSE OS,” select “Use Custom,” and choose the FullPageOS image file.
   - **Choose Storage**: Click “CHOOSE STORAGE” and select your microSD card.

4. **Configure Wi-Fi in Advanced Options**:
   - Press `Ctrl + Shift + X` (or `Cmd + Shift + X` on macOS) to open the Advanced Options.
   - **Enable Wi-Fi**: Check “Configure wireless LAN.”
   - **Enter Network Details**: Input your Wi-Fi **SSID**, **password**, and **Wi-Fi country**.

5. **Write the OS to the microSD Card**:
   - Click “WRITE” to flash FullPageOS with Wi-Fi settings. Eject the microSD card when complete.

6. **Insert the microSD Card and Boot**:
   - Insert the card into the Raspberry Pi, connect it to a monitor and keyboard, and power it on. The Pi should automatically connect to Wi-Fi.

---

### Step 2: Identify the Display Output Name Using `xrandr`

1. **List Connected Displays**:
   - Open a terminal on your Raspberry Pi and run:
     ```bash
     xrandr --listmonitors
     ```

2. **Note the Display Name**:
   - The output might look like:
     ```plaintext
     Monitors: 1
      0: +HDMI-2 1920/527x1080/296+0+0  HDMI-2
     ```
   - Write down the display name (e.g., `HDMI-2`). If it’s different, such as `HDMI-1`, use this name in the `rotate.sh` script.

---

### Step 3: Create and Edit the `rotate.sh` Script

This script allows flexible rotation by accepting an argument (e.g., `left`, `right`, `normal`, `inverted`).

1. **Create the `rotate.sh` Script**:
   - In the terminal, create the `rotate.sh` file in the `scripts` folder:
     ```bash
     sudo mkdir -p /home/pi/scripts
     sudo nano /home/pi/scripts/rotate.sh
     ```

2. **Add the Advanced Rotation Script**:
   - Paste the following code into `rotate.sh`, replacing `HDMI-2` with the correct display name if needed:
     ```bash
     #!/bin/bash
     # This script rotates the display. Specify [normal|inverted|left|right] as an argument.

     if [ -z "$1" ] ; then
       echo "Usage: $0 [normal|inverted|left|right]"
       echo " "
       exit 1
     fi

     function do_rotate {
       xrandr --output HDMI-2 --rotate $1

       TRANSFORM='Coordinate Transformation Matrix'
       POINTERS=$(xinput | grep 'slave  pointer')
       POINTERS=$(echo $POINTERS | sed s/↳\ /\$/g)
       POINTERS=$(echo $POINTERS | sed s/\ id=/\@/g)
       POINTERS=$(echo $POINTERS | sed s/\ \\\[slave\ pointer/\#/g)
       
       iIndex=2
       POINTER=$(echo $POINTERS | cut -d "@" -f $iIndex | cut -d "#" -f 1)
       
       while [ "$POINTER" != "" ]; do
         POINTER=$(echo $POINTERS | cut -d "@" -f $iIndex | cut -d "#" -f 1)
         POINTERNAME=$(echo $POINTERS | cut -d "$" -f $iIndex | cut -d "@" -f 1)

         if [ "$POINTER" != "" ]; then
           case "$1" in
               normal)
                 [ ! -z "$POINTER" ] && xinput set-prop "$POINTER" "$TRANSFORM" 1 0 0 0 1 0 0 0 1
                 ;;
               inverted)
                 [ ! -z "$POINTER" ] && xinput set-prop "$POINTER" "$TRANSFORM" -1 0 1 0 -1 1 0 0 1
                 ;;
               left)
                 [ ! -z "$POINTER" ] && xinput set-prop "$POINTER" "$TRANSFORM" 0 -1 1 1 0 0 0 0 1
                 ;;
               right)
                 [ ! -z "$POINTER" ] && xinput set-prop "$POINTER" "$TRANSFORM" 0 1 0 -1 0 1 0 0 1
                 ;;
           esac
         fi
         iIndex=$((iIndex + 1))
       done
     }

     do_rotate $1
     ```

3. **Save and Exit**:
   - Press `Ctrl + X`, then `Y`, and `Enter` to save the script.

4. **Make the Script Executable**:
   ```bash
   chmod +x /home/pi/scripts/rotate.sh
   ```

---

### Step 4: Test the Rotation Script Manually with `DISPLAY=:0`

1. **Run the Script with `DISPLAY=:0`**:
   - To test the rotation, run the following command, specifying `DISPLAY=:0`:
     ```bash
     DISPLAY=:0 /home/pi/scripts/rotate.sh left
     ```

   - Replace `left` with your desired rotation direction (`left`, `right`, `normal`, or `inverted`).

2. **Confirm the Rotation**:
   - Verify that the display rotates as expected. If it works correctly, proceed to the next step to set up the script to run automatically on startup.

---

### Step 5: Set Up the Rotation Script to Run on Startup with `systemd`

1. **Create a `systemd` Service File**:
   - Open a new service file:
     ```bash
     sudo nano /etc/systemd/system/rotate_screen.service
     ```

2. **Add Service Configuration with `DISPLAY=:0`**:
   - Add the following configuration to the service file:
     ```ini
     [Unit]
     Description=Screen Rotation Script
     After=graphical.target

     [Service]
     ExecStart=/home/pi/scripts/rotate.sh left
     Environment="DISPLAY=:0"
     User=pi
     Restart=on-failure

     [Install]
     WantedBy=multi-user.target
     ```

   - Replace `left` in `ExecStart` with your desired orientation (`left`, `right`, `inverted`, or `normal`).

3. **Save and Exit**:
   - Press `Ctrl + X`, then `Y`, and `Enter`.

4. **Enable and Start the Service**:
   - Enable the service to start at boot:
     ```bash
     sudo systemctl enable rotate_screen.service
     ```

5. **Test the Service**:
   - Start the service immediately to verify it works:
     ```bash
     sudo systemctl start rotate_screen.service
     ```

6. **Check Status**:
   - Confirm that the service is active:
     ```bash
     sudo systemctl status rotate_screen.service
     ```

---

### Final Step: Reboot and Verify Everything

1. **Reboot the Raspberry Pi**:
   ```bash
   sudo reboot
   ```

2. **Confirm Rotation on Startup**:
   - After rebooting, verify that the screen rotates automatically as specified by the `rotate_screen.service`.

---

guysoft, 2024-11-02:
No reply from op, closing for now

closed 2024-11-02 by guysoft

jarsiv, 2024-11-27:
xrandr --listmonitors not working with Raspberry Pi3 and 7" official display. Got error:
Can't open display

Well DSI-1 is working as display name.

but when testing DISPLAY=:0 /home/pi/scripts/rotate.sh, left, right and normal are working with error message:
/home/pi/scripts/rotate.sh: line 14: xinput: command not found

But inverted is not working. Any idea what is wrong with inverted?

guysoft, 2024-11-27:
It looks like the xinput command is used here and is missing:
https://github.com/guysoft/CustomPiOS/blob/feature/meta-modules/src/modules/gui/filesystem/opt/scripts/rotate.sh#L18

Try installing it and running again:
```
apt-get install xinput
```

If it solves the issues I will add it in the gui module to be installed

jarsiv, 2024-11-27:
> @jarsiv It looks like the xinput command is used here and is missing: https://github.com/guysoft/CustomPiOS/blob/feature/meta-modules/src/modules/gui/filesystem/opt/scripts/rotate.sh#L18
> 
> Try installing it and running again:
> 
> ```
> apt-get install xinput
> ```
> 
> If it solves the issues I will add it in the gui module to be installed
> 
> (Needs to be added here https://github.com/guysoft/CustomPiOS/blob/feature/meta-modules/src/modules/gui/start_chroot_script#L20 )

Error message dissapear. Right, left and normal are working pointer and picture are rotating.
With inverted the pointer is only inverted but picture remains as normal.

guysoft, 2024-11-27:
@jarsiv So it does solve your issue?

jarsiv, 2024-11-27:
@guysoft 
No, it didn't solve it. 
So with inverted command, the picture will not turn 180 degree but the pointer will. Picture remain as 0 degree "normal".

It is strange that right, left and normal rotations are working as they should.
I wonder does xrandr has something with this issue. due xrandr --listmonitors gives the error: Can't open display

xrandr seems to be available in source x11-xserver-utils, but that is already installed

jarsiv, 2024-11-27:
@guysoft
I solve this for now with adding the below command in cmdline.txt 
video=DSI-1:800x480M@60,rotate=180 fbcon=rotate:2

Only remain issue is that the pointer is moving correctly but the pointer is upside-down :)

guysoft, 2024-11-27:
Might be worth adding that to the FAQ @jarsiv :
https://github.com/guysoft/FullPageOS/wiki/FAQ#how-to-rotate-the-screen-to-a-vertical-position